SebKuzminsky/pycam #1: Switch from GTK2 to GTK3

njh, 2017-02-16:
Using PyGObject:

> Python Bindings for GLib/GObject/GIO/GTK+
> PyGObject is a Python extension module that gives clean and consistent access to the entire 
> GNOME software platform through the use of GObject Introspection.

https://wiki.gnome.org/action/show/Projects/PyGObject

Debian package: python-gi

njh, 2017-02-16:
On 2017-02-15 12:54, Reuben Rissler wrote:
> I personally would vote to move to GTK3. I have been helping out on 
> StackOverflow for a year now and every time a question is posted about 
> GTK2 the guy is reminded GTK2 is no longer supported. Glade is a drag 
> and drop window designer for GTK3 that makes windows, buttons and so 
> forth fairly easy. I have to check the crossplatform details of GTK3 but 
> I know people are using it on Windows and Linux.

njh, 2017-02-20:
I am a user of MusicBrainz Picard, which is written in Python and works well on Linux/Mac/Windows:
https://github.com/metabrainz/picard

It uses Qt4:
https://riverbankcomputing.com/software/pyqt/intro

And has support for a OpenQL widget:
http://pyqt.sourceforge.net/Docs/PyQt4/qglwidget.html

Looks well maintained...

ebo, 2017-02-28:
Shouldn't this be marked as a feature request instead of an issue?

laconc, 2017-06-28:
The current version is on GTK3; this issue has been addressed.

sumpfralle, 2017-06-28:
Good reminder!

closed 2017-06-28 by sumpfralle