SebKuzminsky/pycam #2: Rescue old Wiki documentation

njh, 2017-02-16:
For example:
http://sourceforge.net/apps/mediawiki/pycam/index.php?title=User_Manual

SebKuzminsky, 2017-02-16:
I don't have a copy of the old wiki, but I fetched the 'pycam-media' repo from SF (https://sourceforge.net/p/pycam/media/)  and pushed it to this pycam repo as a branch called 'media'.

njh, 2017-02-20:
I am putting the new website/documentation in a separate repo, for now.
We can merge the two, if it is a good idea in the future.

All the wiki pages as Markdown:
https://github.com/njh/pycam-website/

Test deployment here:
http://pycam.aelius.com/

sumpfralle, 2017-02-20:
This is a great start!

njh, 2017-02-21:
I am calling this complete. I have rescued and re-integrated the images and cleaned up some of the Markdown.

Any more work on this isn't about rescuing the old documentation.

I think work should continue in #21

closed 2017-02-21 by njh

SebKuzminsky, 2017-02-21:
That's great!