SebKuzminsky/pycam #3: Replace psyco with PyPy

njh, 2017-02-16:
pycam current has a dependency upon psyco.

> "Psyco is a historical project and only works on Python <= 2.6 on 32-bit x86 Intel CPUs."

https://pypi.python.org/pypi/psyco

> Psyco is unmaintained and dead. Please look at PyPy for the state-of-the-art in JIT compilers for Python.

http://psyco.sourceforge.net/

njh, 2017-02-17:
At the moment running without psyco displays the following:
> Psyco is not available (performance will probably suffer slightly)

sumpfralle, 2017-03-05:
I removed all references to `psyco` in order to get rid of the info message.

sumpfralle, 2017-11-11:
If someone is interested in using pycam with the pypy intepreter, please open another issue.

closed 2017-11-11 by sumpfralle