SebKuzminsky/pycam #5: Refactor to support Python 2.7+ and 3.4+

njh, 2017-02-16:
On 2017-02-14 20:09, EBo wrote:
> That said, if others were willing to pick up various balls and carry on 
> I could probably port the Py 2.6 to 2.7 or 3.* isim's.  If we were 
> careful we could get it to work with 2.7+ and 3.4+.  That is about the 
> extent of the time I could break away.

SebKuzminsky, 2017-02-16:
The tip of master currently runs just fine on Python 2.7.

I'm no Python expert, but I'd expect that supporting both Python 2 and Python 3 in a single script would be very hard.

I'm totally fine with Python 2.  I'm in no rush to move to Python 3.

njh, 2017-02-16:
I didn't realise that it master already works Python 2.7.
I will give it a go tomorrow.

I would also be happy to stick with Python 2 for now.

ebo, 2017-02-17:
I've ported a number of programs from Python 2.7 to also work with Python 3.*.  It is not that difficult actually.  There are a couple of things that you have to know, mainly:

Python 3+  requires that all your print statements to have parens around them (was optional in 2.*).  Also, you cannot use some really fun features of 3.4 like full unicode support so that you can use the Greek letter pi for the mathematical constant pi, etc.  I could probably help with that as well.

I would want to wait until the 0.6.0 stabilization release is done before attempting....

laconc, 2017-06-28:
The program successfully runs on Python 3 (in Linux) now, as evidenced by https://github.com/SebKuzminsky/pycam/pull/63. Seems like this issue can be closed.

sumpfralle, 2017-06-28:
Agreed.

closed 2017-06-28 by sumpfralle