SebKuzminsky/pycam #10: Get pycam into Homebrew

njh, 2017-02-17:
- [ ] build homebrew Formula
- [ ] get second person to test
- [ ] submit to Homebrew
- [ ] update pycam documentation for Mac

njh, 2017-02-18:
Thought I was close to getting it working and then found I could get gtkgl to work:

```
$ pycam
Psyco is not available (performance will probably suffer slightly)
Enabled 4 parallel local processes
Fatal Python error: can't initialize module gtk.gtkgl
Abort trap: 6
```

It doesn't look like there has been any work on it for 7 year, so so not feeling optimistic about getting it to work on Mac:
* https://github.com/GNOME/pygtkglext
* https://projects.gnome.org/gtkglext/download.html#pygtkglext

sumpfralle, 2017-02-18:
I think, the topic "pycam + Mac" was the most frequently asked on in the pycam forum on sourceforge. Back then it always seemed to take a lot of effort (but worked somehow).
Given that this was some years ago, personally I would defer any thoughts about running pycam on Mac to the time after switching to another visualization software and maybe even after switching to gtk3 or something else.

njh, 2017-02-18:
Yes, I think I am inclined to agree with you. Might take too much to fix time in the short-term 😕

njh, 2017-02-20:
My draft Homebrew Formula (doesn't currently work):
https://gist.github.com/njh/960050f42111fa66718fd76a3d7ce2dd