SebKuzminsky/pycam #15: Guppy dependency

valeriob01, 2017-02-18:
The Memory Analyzer plugin requires the Guppy module which is not installed by default.

njh, 2017-02-18:
Hm, there doesn't seem to be a Debian package for Guppy.

But there is for Ubuntu:
* http://packages.ubuntu.com/trusty/python-guppy

sumpfralle, 2017-02-19:
The MemoryAnalyzer plugin is only useful for debugging and development.
Pycam handles the absence of guppy gracefully.
Do you think, that something needs to be done?

SebKuzminsky, 2017-02-19:
python-guppy is currently Recommended by the pycam debian package, which means installing pycam will install guppy if it's available, but won't prevent installation of pycam if guppy is not available.  I think that's the best we can do for now.

If Debian adds python-guppy in the future we can promote it from Recommends to Depends.

closed 2017-02-19 by SebKuzminsky