SebKuzminsky/pycam #16: Setup Travis

njh, 2017-02-18:
I have just pushed a basic .travis.yml file:
https://github.com/SebKuzminsky/pycam/blob/master/.travis.yml

Can you enable it please @SebKuzminsky?

(fingers crossed the job won't fail!)

Future things to add?
* pylint
* Some kind of simple 'smoke test' - run pycam with command line arguments?
* Test building the Debian package?
* Add unit tests

sumpfralle, 2017-02-19:
That looks good!
I just added the pyflake8 style check (less pedantic than pylint). pylint should not be required / suitable for now.

btw: I noticed, that you (njh) used multiple (slightly confusing) merge operations for your commits. I guess that you accidentally used `git pull` instead of `git pull --rebase` while working directly on the master branch? Maybe it would be easier for you to work on a separate branch.

Regarding command line operations: previously there were command line operations. But I did not move them to the new internal structure after v0.5.1. Thus we cannot use these for tests at the moment.

There are already some unittests (pycam/Test), but they fail currently. I think, I will take a look at them now.

SebKuzminsky, 2017-02-19:
I'm super excited about the command-line interface.

njh, 2017-02-19:
I agree that having a command line interface would be great.
Makes it possible to automate g-code generation :)

But for a smoke test, even just running with --help would check that things aren't complete broken.
And useful for checking pull requests.

SebKuzminsky, 2017-02-19:
We run with --help as part of building the manpage.

SebKuzminsky, 2017-02-20:
Travis now runs, so I'm closing this issue.

https://travis-ci.org/SebKuzminsky/pycam/

I added "pycam --help" to the travis job, this currently demonstrates a problem, new since 0.6.

I also added a Travis pass/fail icon to the top of our README.

closed 2017-02-20 by SebKuzminsky

sumpfralle, 2017-02-21:
I switched the style checker now back to `flake8` by using a wrapper script. It looks like there is no other way :(
@SebKuzminsky: thank you for your thorough explanation of the problem regarding flake8 on different platforms. That helped a lot, since I do not have an Ubuntu system around.

Now there seems to be a problem with an (currently unconditional) import of OpenGL.

SebKuzminsky, 2017-02-21:
Yeah, I'm chasing that OpenGL issue right now.  I guess i don't really understand the Travis environment yet...

SebKuzminsky, 2017-02-21:
Here's why Travis can't find OpenGL (or any other python packages we install via apt):

https://docs.travis-ci.com/user/languages/python/#Travis-CI-Uses-Isolated-virtualenvs

SebKuzminsky, 2017-02-21:
Ok, with e476593 I'm confident it'll find OpenGL and the other python packages we got from apt.

But now we're back to a problem with flake.

SebKuzminsky, 2017-02-21:
And we have a passing build!