SebKuzminsky/pycam #18: Step Down doesn't work for Engrave

njh, 2017-02-19:
I am currently using pycam to cut a 2D shapes out of 12mm plywood.

* In the Processes section I have set the Step Down to 3mm.
* In the bounds section, I have set lower Z to 12mm

But in the generated G-code, it goes down to Z-12 in a single step.

Pretty sure this used to work. I will try and pick my way though the code and see if I can work out what is wrong.

sumpfralle, 2017-02-22:
I tried the same some minutes ago and I noticed the same: the cutter goes down to the model in one step.

Later I realized, that I had set the lower z to 12mm - but I should have used the upper z limitation instead.
The cutter always starts with the upper z level and steps down to the lower z level or the model plane (whichever is higher). In my (wrong) setup the upper z margin was zero - thus only one step was required, since the distance was zero.

If your test was flawed in the same way as mine (mixing up lower and upper z margins), then there is not code error, but there is surely a usability problem. In this case we should try to visualize the six margins (two for each axis) in a more unambiguous way.

njh, 2017-02-22:
Ah! That could very well be it.

I found setting the bounds a very confusing way of setting the depth for 2D engraving.
There has to be a simpler/easier to understand way of doing it.

I wondered about setting the engrave depth in the same place as Step Down - in the processes section - but I am not sure if that impact more advanced usage.

sumpfralle, 2017-02-23:
Let us start a discussion about the interface / setting on the mailing list?

valeriob01, 2017-03-02:
I have set lower and upper bounds right, still it ignores the step down.

sumpfralle, 2017-03-02:
For me it seems to work with the current state of the repository:
[image]

valeriob01, 2017-03-03:
I have found, the bounds tab is happier if you set bounds in mm/inch rather than percentile.
By setting mm/inch it works for me also. When setting percentile it does not work and the Z size is 0.

valeriob01, 2017-03-03:
Probably it is just me being dumb, I always set the lower bound to -something and the upper bound to 0. I will try to set an upper bound of +something.

valeriob01, 2017-03-03:
Ok, I have done so and this way it works. If I set the z lower bound to a negative value it does not work.

ebo, 2017-03-04:
I see a couple of different issues here.

*) the use of % -- no matter what your upper and lower limits are set to, specifying % and/or specific values should just work.  I would guess that there is a bug in how the % works.  You rpesent some interesting boundry cases and we should set up a test cases.

*) the use of defining upper and lower z -- there should be some sanity checking to make sure that the upper > lower, and upper <= lower should at least issue a warning if not an error.

There is also an issue of how to interpret the plunge.  One way is to define the path as lowering down to some % of depth and make the first run around the edge (sumpfralle's example above).  Another would be to peck drill down to depth and then do a single pass around.  The last is to simply feed down to depth and trace.

I guess that I see both % and hard coded values as valid, but we have to carefully define the expected behavior.

valeriob01, 2017-03-05:
Test: the model is positioned at Z -2.0
[image]

Note: I have set a Z margin of 10%, the Z size is 0
[image]

The toolpath plunges down to -2 even the step down is set to 1.0
[image]

[image]

This does not happen if the margins are set in mm/inch.

sumpfralle, 2017-03-05:
Within the logic of the program this is just as expected: the model's z dimension is zero. Thus a 10% margin adds just zero.
You see the result in the bounding box dimension in the head of the visualization window (`Bounding Box: ... deltaZ=0`).
Code-wise everything is fine. We just need to design a less surprising GUI.

valeriob01, 2017-03-05:
At a minimum a revised UI could check if the user specifies a step down that is different from the Z size.

sumpfralle, 2017-03-05:
The step-down value is just the distance between layers. If the user wants just one layer (dimension of bounding box for z being zero), this should be possible. Maybe a warning could appear (similar to the one that appears in the status bar if a toolpath was generated without selecting a collision model).
Anyway: we should probably first decide about the internal logic (being discussed on the mailing list) and then about the corresponding GUI.
It will get better ...

ebo, 2017-03-05:
If it is consistent internally, then it is mainly a UI issue.

On Mar 5 2017 7:56 AM, sumpfralle wrote:
> Within the logic of the program this is just as expected: the model's
> z dimension is zero. Thus a 10% margin adds just zero.
> You see the result in the bounding box dimension in the head of the [...]

valeriob01, 2017-03-05:
On Sun, 2017-03-05 at 07:27 -0800, sumpfralle wrote:
> The step-down value is just the distance between layers. If the user wants just one layer (dimension of bounding box for z being zero), this should be possible. Maybe a warning could appear (similar [...]

Yes, currently the only way to make more passes is to specify a Z
thickness and a step down, which is logically good, it is not possible
to specify the thickness in % but then it can be specified in mm/inch.