SebKuzminsky/pycam #19: Error while starting

valeriob01, 2017-02-20:
:~/Downloads/pycam/scripts$ ./pycam
Traceback (most recent call last):
  File "./pycam", line 41, in <module>
    from pycam import VERSION
ImportError: No module named pycam

valeriob01, 2017-02-20:
None of the tests run also with the same error ImportError:

~/Downloads/pycam/Tests$ ./PushCutterTest.py 
Traceback (most recent call last):
  File "./PushCutterTest.py", line 22, in <module>
    from pycam.Geometry import Point
ImportError: No module named pycam.Geometry

njh, 2017-02-20:
This is also the reason why the Travis build is currently broken:
https://travis-ci.org/SebKuzminsky/pycam

SebKuzminsky, 2017-02-20:
"pycam --help" worked in 0.6 and broke in master on the commit where sumpfralle-style-pep8 was merged.

I bisected sumpfralle-style-pep8 and discovered that 1df9adf4c850547d5431cb37e369ae45e1b80d4d is the first commit where "pycam --help" doesn't run, though the error message is different than what we get now:

```
Traceback (most recent call last):
  File "./scripts/pycam", line 39, in <module>
    import pycam.Gui.Settings
  File "/home/seb/cad-cam/pycam/pycam/Gui/Settings.py", line 23, in <module>
    from pycam.Toolpath import Bounds
  File "/home/seb/cad-cam/pycam/pycam/Toolpath/__init__.py", line 29, in <module>
    from pycam.Geometry.PointUtils import *
  File "/home/seb/cad-cam/pycam/pycam/Geometry/__init__.py", line 26, in <module>
    import pycam.Geometry.Matrix as Matrix
AttributeError: 'module' object has no attribute 'Geometry'
```

SebKuzminsky, 2017-02-20:
The error changed to what we have now after commit 32727e6e36dbb6300ee86cd7ee0fec9112b31fae:

```
Traceback (most recent call last):
  File "./pycam", line 47, in <module>
    from pycam import VERSION
ImportError: No module named pycam
```

The only suspicious-looking thing in that commit is the move of the line `from pycam import VERSION` to come before all the other importing of pycam stuff.

Moving that line to the end fixes the `ImportError: No module named pycam` problem, revealing the `AttributeError: 'module' object has no attribute 'Geometry'` problem again.  I did this change in commit 070ebe5 just now.

I bet it's some pythonic confusion about whether "pycam" means the script pycam or the module pycam, and how that relates to importing, and some interaction with the precedence rules of importation.

closed 2017-02-21 by sumpfralle

sumpfralle, 2017-02-21:
I also suspected a name collision due to the script name (`pycam`) - but I could not confirm this.

When running pycam locally it is required to add the project directory to the PYTHONPATH. Now this is done (again) in the start script (`scripts/pycam`).

commit a12fa9a1 (2017-02-21): start script: manually set PYTHONPATH

Closes: #19