SebKuzminsky/pycam #21: Revive/redirect/shutdown the project website

sumpfralle, 2017-02-21:
Currently it is located here:
 http://pycam.sf.net/

Its content is stored in a git repository (https://sourceforge.net/p/pycam/media/ci/master/tree/website/). It can be updated by members of the sourceforge project. Access can be given to whomever wants to work on it.

What should we do about it?
* Update the few outdated links and show some signs of life?
* Replace the site with a redirect to the (not yet final) github project?
* ???

njh, 2017-02-21:
I propose doing a little bit more work on the documentation site I have been working on:
http://pycam.aelius.com/

And then uploading those pages to http://pycam.sf.net/

njh, 2017-02-22:
Now live: http://pycam.sourceforge.net/

Lots more to do but we have a new site up and running now.

One final thing before I close this ticket. Where should the website markdown files and config live?

1. In its own special repo (as it is now) - many projects, including LinuxCNC, do this
2. In its own orphan branch in the main repo
3. In the same branch as the code

I think option 3 is my preference - the markdown files readable on GitHub and also as plain text files. The markdown files and images currently use up 1.7Mb of space.

sumpfralle, 2017-02-23:
This looks great!

I would also prefer options 3 for now.

njh, 2017-02-23:
Ok, I have gone for option 3 and it is in the master branch now.

Relevant files are:
* https://github.com/SebKuzminsky/pycam/blob/master/mkdocs.yml
* https://github.com/SebKuzminsky/pycam/tree/master/docs

To run local webserver, install [mkdocs](http://www.mkdocs.org/) and run:

    mkdocs serve

To build and upload the documentation to pycam.sf.net run:

    upload-docs

I will put this in the Developer Documentation at some point.

closed 2017-02-23 by njh