SebKuzminsky/pycam #22: G4 after M3 should be optional, the default should be "no"

SebKuzminsky, 2017-02-22:
Currently in G-code exported by PyCAM, M3 is followed by G4 P3.

M3 starts the spindle.  G4 pauses for P seconds.

It's important to not start cutting before the spindle is up to speed.  Well-wired machines know when the spindle is up to speed via closed-loop feedback.  Poorly wired machines don't have spindle speed feedback.

So PyCAM is targetting a poorly wired machine, and furthermore hoping that 3 seconds is enough time for the spindle to come up to speed.  A better default would be to assume that the target machine has spindle speed feedback, so that G4 is not needed.  Let the user provide a G4 delay if they want, and let them pick the spin-up time, but omit it by default.

valeriob01, 2017-02-22:
> Currently in G-code exported by PyCAM, M3 is followed by G4 P3.
>
> M3 starts the spindle.  G4 pauses for P seconds.
>
> It's important to not start cutting before the spindle is up to speed.  Well-wired [...]

This is good, a poorly wired machine could have spindle with manual
on/off, so the M3 has no effect :-)

sumpfralle, 2017-02-23:
My (surely poorly wired) machine's spindle is turned on via M3 and there is no closed-loop feedback. But I still like this machine :)

Personally I would prefer to stick to safe defaults. In this case it would be the longest period of time a reasonable machine (whatever that means) requires for starting the spindle.
Users of such poor machines will be happy that their tool survived the first run with pycam-generated GCode.
Users of well-wired machines will notice the delay and look for the respective setting of the GCode generator in pycam.

Thus I would argue that we should default to preventing the machine from being damaged.
(instead of phrasing it "default to poorly wired machines")

But I (as a machinist) have very limited knowledge of the world and common machines. Thus you may feel free to ignore my thoughts.

valeriob01, 2017-02-23:
Hello,
my machine has a vfd with manual on/off, I have generated a moderately
big amount of gcode with pycam, each time forgetting to deactivate the
M3, it was however safe to do so, as the M3 has no effect at all, the
only noticeable difference is the delay at start, but this is safe too.
So yes, I would stick with the current safe defaults.
However, I would also make the G4 optional.

SebKuzminsky, 2017-02-24:
Ok after this discussion I changed the title of this issue, the new goal is:

Preserve current behavior (M3 followed by G4P3), but provide a way for the user to override the G4 P-word, or disable it entirely.

sumpfralle, 2017-02-25:
I just noticed, that this feature request describes the current state of v0.6 :)
Hidden in the `Toolpaths -> GCode -> Settings` dialoag there is the spindle delay setting. Setting it to zero causes the G4 line to disappear.

Should we do something about the usability (e.g. how to find it) of this option? Or is it fine as it is?

SebKuzminsky, 2017-02-25:
I just noticed the same thing.  This issue was just me being dumb.

closed 2017-02-25 by SebKuzminsky