SebKuzminsky/pycam #23: How can we Benchmark PyCam?

njh, 2017-02-22:
Models
------

### Model types

-   a 3D model mainly consisting of planes and cylinders
-   a 3D model with an organic (non-flat) shape

### Model properties

-   around 500 triangles
-   dimension: 30 x 30 x 15 mm

Processes
---------

### Slice removal / PushCutter

-   line distance: 3mm
-   step-down: 3mm
-   step-forward: not required for PyCAM (but required for others?)

### Contour (3D)

-   step-down: 3mm
-   ??? further attributes depend on the implementation ???
    -   line distance (for a grid-based approach - see PyCAM's old
        Contour strategy)

### DropCutter

-   line-distance: 1mm
-   step-forward: not required for PyCAM (but required for others?)

Tools
-----

-   flat end / ball nose / bull nose
-   diameter: 3mm

Potential problems
==================

-   how can we compare the quality of the result?
-   resources: single core / multi-core / cloud processing ?
-   different implementations require different parameters

closed 2017-02-22 by njh

sumpfralle, 2017-02-23:
Probably the task was closed automatically due to being mentioned in the commit text?

njh, 2017-02-23:
I made a ticket to make sure it was captured somewhere but closed it to avoid clutter - re-open if you think we should address it sooner rather than later.