SebKuzminsky/pycam #24: Specify feedrate for plunge move

njh, 2017-02-23:
_From the old MediaWiki_

Different materials and different tools are more or less suitable for moving straight downward into material quickly. This specific feedrate should be configurable.

valeriob01, 2017-02-26:
This is a really wanted feature. Said in other words: it is a limitation that forced me to correct the generated code by hand (adding the plunge feedrate by hand).

njh, 2017-02-26:
Me too - I have been manually editing after plunging too fast and breaking a tool.

sumpfralle, 2017-02-26:
In GCode terms: what would be the best approach for doing this?

For now I assume the following.
Whenever there is a `G1` move heading towards a lower Z level, then:

1. reduce the feedrate according to the given angle and the maximum plunge rate (`F...`)
2. execute the move (`G1 ...`)
3. restore the previous feedrate (`F...`)

Or are there other gcode commands allowing to change the feedrate for a single line, only?

valeriob01, 2017-02-26:
Not that I know of. But you can terminate each line with F...

sumpfralle, 2017-02-26:
@valeriob01: but this would change the feedrate not only for the current line but also for all following lines, or?

valeriob01, 2017-02-26:
I think the F word at the end of line changes the feedrate only for the current line, you can test this:

F100
G01 X0.0 Y0.0 F200
G01 X10.0 Y0.0 F200
G01 X0.0 Y0.0
G01 X10.0 Y0.0

and look how it behaves

sumpfralle, 2017-02-26:
This inline-style would be perfect.

Regarding the test: I would rather like to follow a specification and not the behaviour of an implementation :)

Here is the gcode documentation of linuxcnc regarding [G1 and feed rate](http://www.linuxcnc.org/docs/html/gcode/g-code.html#_g1_linear_move).
For me it is not clear, that this inline feed rate specification is valid only for this specific line. But maybe it is just me being a non-native English speaker?

valeriob01, 2017-02-26:
On Sun, 2017-02-26 at 06:49 -0800, sumpfralle wrote:
> This inline-style would be perfect.
>
> Regarding the test: I would rather like to follow a specification and not the behaviour of an implementation :)
>
> Here is the gcode documentation of linuxcnc [...]

I found this example in the linuxcnc docs:

G1 Example

G90 (set absolute distance mode)
G1 X1.2 Y-3 F10 (linear move at a feed rate of 10 from current position
to X1.2 Y-3)
Z-2.3 (linear move at same feed rate from current position to Z-2.3)
Z1 F25 (linear move at a feed rate of 25 from current position to Z1)
M2 (end program)

That is: if you omit the F word, the feedrate is that of the last F
word! so I was wrong. The last F word is the valid one.

valeriob01, 2017-02-26:
There is at least one case where the plunge rate should be the 'fast' one: the case where G02/G03 specifies a Z -xxx, in this case the plunge move is interpolated, it doesn't plunge vertically but rather it makes a spiral down the Z axis.

sumpfralle, 2017-02-26:
G02/G03 would probably be similar to the case of G01 if the move is not strictly vertical, but has a certain angle. Here I thought that I would calculate the feed rate in a way, that it the Z direction feed rate does not exceed the plunge speed limit.

Or where you thinking only of straight vertical Z moves?

valeriob01, 2017-02-26:
On Sun, 2017-02-26 at 11:59 -0800, sumpfralle wrote:
> G02/G03 would probably be similar to the case of G01 if the move is not strictly vertical, but has a certain angle. Here I thought that I would calculate the feed rate in a way, that it the Z direction feed rate does not exceed the plunge speed limit.

yes, for example when I cut aluminum, I use a plunge rate that is 1/10
of the linear move rate. To make a spiral circle in aluminum I use for
example:

G02 X0.0 Y0.0 Z-0.5 I0.0 J50.0 F400

while for a vertical plunge I use G01 Z-0.5 F40

sumpfralle, 2017-02-26:
OK - we are on the same track.

Regarding G02/G03: these are not created by PyCAM for now. Thus this would be a thought for later.

commit 08792198 (2017-02-28): limit the feedrate of downward moves

An additional GCode setting allows to limit the feedrate of (partial or
fully) downward moves. This allows to use different feedrates for
horizontal and vertical moves.

Closes: #24
closed 2017-02-28 by sumpfralle