SebKuzminsky/pycam #25: Support for opencamlib

njh, 2017-02-23:
_From the old MediaWiki WantedFeatures page_

[opencamlib](http://code.google.com/p/opencamlib/) ...

> ... a GPL licensed C++ library with Python bindings for creating toolpaths for cnc-machines.

*opencamlib* currently supports some features that are not available in
PyCAM:

-   waterline path with adaptive accuracy
-   voronoi diagrams (to be used for polygon offsetting)
-   better performance

Tasks:

- [ ]  develop some python modules for interfacing with opencamlib (partly already done on their side)
- [ ]  basically: how to convert a PyCAM 3D model to an opencamlib 3D model (and the other way around for the resulting toolpath)
- [ ]   integrate opencamlib as a new separate backend (currently: triangle and ODE)
- [ ]   update the Debian/Ubuntu/??? usage instructions
- [ ]   integrate opencamlib into the windows package builder setup (pyinstaller)

TurBoss, 2017-05-26:
maybe this could help
https://github.com/vilemduha/blendercam/tree/master/scripts/addons/cam/opencamlib

sumpfralle, 2017-05-26:
This looks like a good code example. Thank you for sharing it!