SebKuzminsky/pycam #27: Projecting engravings onto 3D models

sumpfralle, 2017-02-23:
t would be nice to be able to project a planar engraving (e.g. from a DXF file) onto a trimesh model (e.g. from an STL file). Combining the EngraveCutter (following lines) with a DropCutter (following heights) and moving each point of the resulting toolpath down the z axis should do the trick.

thawkins, 2017-04-10:
would this allow me to cut profiles out from acrylic, i need to execute the same outline toolpath multiple times, but drop it down by a small amount on each pass, I can create stl files for my profiles in openscad, but i cant work out how to generate the required path in pycam. 

I have attached an example of what I want to do (example from grblSender) 

[electric_turtle.zip](https://github.com/SebKuzminsky/pycam/files/908941/electric_turtle.zip)

sumpfralle, 2017-04-10:
This specific issue here discusses the projection of 2D models onto (non-flat) 3D models. I understand you are looking for a simple 2D toolpath instead.

Maybe these articles help you:
* [Input Formats](http://pycam.sourceforge.net/supported-formats/)
* [Processing 2D models](http://pycam.sourceforge.net/modeling-inkscape-svg/) (here: from inkscape)

sumpfralle, 2018-08-20:
It works now:

[image]

closed 2018-08-20 by sumpfralle