SebKuzminsky/pycam #29: Python sandbox for public servers

sumpfralle, 2017-02-23:
The server mode allows execution of arbitrary python code, thus it could be harmful for providers of public PyCAM servers. A kind of sandbox would be helpful. Maybe one of the following projects could be useful:

* http://pypi.python.org/pypi/apipkg
* http://pypi.python.org/pypi/RestrictedPython/
* http://code.google.com/intl/de/appengine/docs/python/runtime.html

njh, 2017-02-24:
Are public servers a good idea at all?

sumpfralle, 2017-02-24:
Probably not.
I just wanted to conserve the thought.

valeriob01, 2017-03-08:
There is a PyPy version that offers sandboxing:
https://pypy.org/download.html#other-versions
https://bitbucket.org/pypy/pypy/downloads/pypy-1.8-sandbox-linux64.tar.bz2