SebKuzminsky/pycam #34: Runtime error while connected to a task server

valeriob01, 2017-02-25:
Command line on the server
--
./pycam --start-server-only --server-auth-key xxxxx --trace

On the client
--
Generated two toolpaths from pycam-textbox.stl,
during a simulation the following error happened while playing with the progress bar

Error
--
Enabled 4 parallel local processes
Allow remote processing
Connected to a remote task server.
Toolpath generation time: 26.587809
Toolpath generation time: 25.972759
Process spawn:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sel/Downloads/pycam/pycam/Utils/threading.py", line 384, in _spawn_daemon
    cache.expire_cache_items()
  File "<string>", line 2, in expire_cache_items
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 773, in _callmethod
    raise convert_to_error(kind, result)
RuntimeError: dictionary changed size during iteration

sumpfralle, 2017-02-25:
is this reproducable? Always? Sometimes?

valeriob01, 2017-02-25:
This was the first time it happened, I will retry tomorrow to reproduce it.

valeriob01, 2017-02-26:
Ok, today I have reproduced the error 2 times. The server crashes with the error reported after a series of messages on the screen:

"Worker thread finished after 60 seconds of inactivity"

sumpfralle, 2017-02-26:
Always? Sometimes? Rarely? :)

valeriob01, 2017-02-26:
It happened 3 times over 4 tests.

sumpfralle, 2017-02-26:
I think, it was related to the expiry of cache items. Should be fixed now.

closed 2017-02-26 by sumpfralle

valeriob01, 2017-02-26:
It seems to be fixed.

commit 54be0e27 (2017-02-26): avoid "changed size during iteration" errors during multiprocessing

Closes: #34