SebKuzminsky/pycam #35: EOF Error

valeriob01, 2017-02-26:
An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot!
Traceback (most recent call last):
  File "/home/sel/Downloads/pycam/pycam/Plugins/Tasks.py", line 317, in generate_toolpath
    toolpath = task_resolver(task, callback=draw_callback)
  File "/home/sel/Downloads/pycam/pycam/Plugins/TaskTypes.py", line 71, in run_task
    maxz=high[2], draw_callback=callback)
  File "/home/sel/Downloads/pycam/pycam/PathGenerators/PushCutter.py", line 89, in GenerateToolPath
    progress_counter)
  File "/home/sel/Downloads/pycam/pycam/PathGenerators/PushCutter.py", line 150, in GenerateToolPathSlice
    for points in run_in_parallel(_process_one_line, args, callback=progress_counter.update):
  File "/home/sel/Downloads/pycam/pycam/Utils/threading.py", line 515, in run_in_parallel_remote
    if not remote_cache.contains(data_uuid):
  File "<string>", line 2, in contains
  File "/usr/lib/python2.7/multiprocessing/managers.py", line 759, in _callmethod
    kind, result = conn.recv()
EOFError

This error after being idle for some time, probably the connection timed out. After reconnecting (without closing the client) the error disappeared.

valeriob01, 2017-02-26:
Just occurred a second time.

sumpfralle, 2017-02-27:
If this would be easily reproducible (within seconds?), it should be possible to find a suitable location for an exception handler. What do you think?

valeriob01, 2017-02-27:
It is reproducible, and it is an error that does not crash the client or blocks in any way, I am not really a python programmer so my advice about would be naif, but I am slowly learning.

sumpfralle, 2017-02-27:
For debugging it is crucial (i.e. "a lot easier"), if the error condition can be triggered arbitrarily.
The above sounds, like you would need to wait for the EOFError exception (seconds, minutes, hours?).

valeriob01, 2017-02-27:
On Mon, 2017-02-27 at 04:54 -0800, sumpfralle wrote:
> For debugging it is crucial (i.e. "a lot easier"), if the error condition can be triggered arbitrarily.
> The above sounds, like you would need to wait for the EOFError exception (seconds, minutes, hours?).

I have updated pycam and put under test by running the client with
--trace option, established connection to task server, waited 3 hours,
and I was unable to reproduce this error.

sumpfralle, 2017-02-27:
Interesting!
In this case I think it was a side-effect of the the two previously fixed issues that you reported (regarding "changed size while iterating"). Whenever one of these issues appeared, I would expect a broken connection (which could cause an "EOFError").
Thus I am closing this, since the root cause seems to be fixed recently.
Thank you for testing it!

closed 2017-02-27 by sumpfralle