SebKuzminsky/pycam #42: Support for VCarving

jtpedersen, 2017-03-04:
V carving is nice for signs and engraving.

sumpfralle, 2017-03-04:
Please tell us more! :)
What is the difference to the current 2D engrave operation?

jtpedersen, 2017-03-04:
VPocket/VCarve is where a V-bit is use and the Z-height is adjusted so it covers a polygon.

Its is often used for lettering, F-engrave is an example (http://www.scorchworks.com/Fengrave/fengrave.html)

Having VCarving within pycam would enable signmaking with more elaborate borders and ornamnets carving.

jscut (http://jscut.org/) has an implementation (which is somewhat limited and had bugs and is abbandonedn)

There are also some examples in opencam lib (at least in openvoronoi), medial_axis

sumpfralle, 2017-03-04:
ah - interesting!

Since PyCAM currently only supports single-line fonts, this would be a indeed different approach.

How would you tackle this problem?
* create the required polygons(?) within PyCAM by using the locally available true-type fonts?
* create the polygons(?) via an external application (e.g. inkscape)?
* V-bit tools are probably not suitable for flat surfaces. Would this require a second tool run?
* there are probably different kind of V-bits around, or? (e.g. curved instead of a straight cone)

btw: PyCAM already includes an "extrude" feature, that turns 2D shapes into curved 3D models. This could achieve a similar outcome, but is surely a huge waste of machining time (using inappropriate tools with inefficient toolpaths).

jscut looks interesting - but my JS reading capabilities are way too poor to draw inspirations from their modeling and processing.

Regarding opencamlib/openvoronoi: if you would like to help with integrating such a function from that library, it would be tremendously great!

jtpedersen, 2017-03-04:
I think the most reasonable approach would be to use openvoroni and i am indeed interested in trying to integrate opencamlib. I am looking at it, but I am most comfortable on the c++ side of things and are not that familar with the c++/python interface.

I think the most flexible KISS solutions is to use external generated polygon. 

There are indeed  issue with flat areas, one of the limitations in jscut is that i just plunges down to infinity without stopping.  A option is to combine an flat endmill and a vbit to cover flat areas, or covering flat areas with mulitple very close paths. It should be possible to inset the polygon to get area that can not be covered by a vcarve and generate a flat endmill path for that..

Using other shaped bits are not suited for a strategy that works on the "medial_axis" as there are more geometry to take into account.

sumpfralle, 2017-03-04:
So if you would like to work from the c++ side, then we can probably join our forces :)

Did you already take a look at pycam/Toolpath/Openvoronoi.py?
This was an attempt of mine to integrate part of openvoronoi for the pocketing functions within pycam. I am not really sure about the state of this code (I did it four years ago and left it in an unclear state), but maybe it gives you a start.

Regarding the polygons: do you know free software, that would create suitable polygons for carving? Would inkscape be sufficient?

Regarding flat areas: for now I would suggest to focus on the edges and not the flat planes. They can be easily done with a second tool afterwards. Later we may want to add the option of using the v-bit also for a rough cleanup of flat areas within a given tolerance (if tool change is too hard for the specific machine).

jtpedersen, 2017-03-05:
I have experimented a bit with the openvoronoi python api i took some inspiration form the pycam/Toolpath/OpenVoronoi.py file.

Got a bit sidetracked on reading Svgs into openvoronoi. The good thin is that I have a SvgReader that can adaptively turn svg paths into polygons, the annoying thing is that i am able to provoke a lot of errors inside openvoronoi. 

Anyhow the SvgReader could perhaps serve as a Importer, the current SvgImporter seems to do a conversion from svg->eps->dxf and then import the dxf

sumpfralle, 2017-03-05:
I cannot remember right now, why I picked this conversion path some years ago.
You may feel free to replace this with a native (external?) SVG parser for polygons.

SebKuzminsky, 2017-03-07:
@jtpedersen, I bet @aewallin would be interested to know about the openvoronoi issues you're running into.

sle118, 2017-09-23:
I'm going to follow this closely; I am new to the CNC world and I have a lot to learn still but it would be really interesting to see this implemented. As @jtpedersen wrote, this additional feature would make pycam a completely integrated solution for engraving on complex shapes!