SebKuzminsky/pycam #44: Engrave offset

fraimondo, 2017-03-09:
I've noticed that in the documentation and the code there is a mention to an "engrave_offset" parameter. The problem is that I did not find it in the UI. There is no way to set a negative engrave offset since the only parameter that changes the offset is the tool diameter.

I've managed to change the [code to add this to the UI](https://github.com/fraimondo/pycam/commit/7b32524d4e0212598d14db11cb999677f666a0aa), but not really a solution, since I don't really understand how the internal works.

Any insights?

sumpfralle, 2017-03-09:
hehe - you were digging deeply in the code! :)
Indeed there was the parameter "engrave_offset" used in previous versions. Somewhen it was turned into an implicit parameter - then it just depended on the tool diameter.
The occurrences of "engrave_offset" in the code are lingering there in some unused pieces of code. I am quite sure, that it is mainly not in use anymore.
I will try to clean this up.
Thank you for taking an intense look!

Were you looking for some specific feature?

fraimondo, 2017-03-09:
Indeed I was trying to cut a shape but compute the exact negative offset.

Sometimes I want a hole to be exactly X mm in diameter, but sometimes I
want to cut out a circle of X mm, so I wanted to invert the offset computed
from the tool diameter.

sumpfralle, 2017-03-10:
Are you sure that the "engrave" parameters are not sufficient for your tasks?
"Pocketing" (holes/material) determines in which direction the tool should go (inside or outside of the polygons).
Or are you looking for something else?

fraimondo, 2017-03-10:
If I select anything on that option, then the struggle with the openvoronoi library starts. And it does something completely different. Although I don't know if it was because I change the code (broken imports + unpacking errors in OpenVoronoi.py).

sumpfralle, 2017-03-10:
Just guessing (as far as I understand you): without the "pocketing" paramter the openvoronoi library is not used at all, since the toolpath simply follows the lines of the polygons.
With pocketing enabled (inwards or outwards) the openvoronoi library comes into play.

I am really unsure, what is the current state of the experimental openvovonoi module in pycam. I started it some years ago and forgot, how it was back then.
Any problems that you encounter are probably based on incomplete code within this python module.

Please share your problems here with us - maybe we can work it out together.

fraimondo, 2017-03-11:
Exactly. With no pocketing the openvoronoi is not used, but I can still use the tool diameter to account for the offset. Weird thing is that I tried to cut two shapes. In one, the offset was on the inside (so the hole in the material gets the dimensions of the path). In the other one, the offset was on the outside (so I cut out a shape exactly like the one in the path). I ended up adding the option to set manually the offset with negative values as an option.

The openvoronoi support is really broken. To make it work, first I fixed some import errors in OpenVoronoi.py. Then some values returned from the openvoronoi library were not as expected. But in the end, I ended up with something unusable: I want to create a 5mm hole and in the end I ended up with something like 10 mm, and there was no overlap between concentric circles. But maybe is because I don’t really get the concept of using this library.

fraimondo, 2017-03-11:
This is an example of what happens when I want to cut a 2cm x 2cm square from Z=0 to Z=-5. Both pocketing options give me the same result.

[image]

sumpfralle, 2017-03-11:
Regarding the inside/outside of polygons: this is determined by the "direction" of the lines. This is really not cool, but I am not aware of a better approach. See the [FAQ](http://pycam.sourceforge.net/faq/#toolpaths-with-an-offset-are-placed-inside-of-the-model-instead-of-outside).

I appreciate, that you phrased it correctly that openvoronoi _support_ (the python module for pycam, that I wrote) is broken. In contrast the openvoronoi library (by Anders Wallin) is probably working properly. Thus we just need to fix our python module.

Regarding your picture: your model is currently located at `z=-4`. Probably you wanted `z=-5`? I am not sure if this is related to your problem.
Besides this: the toolpath in the picture looks good to me. What else did you expect? (maybe I misinterpreted your description)

Could you please push your (probably highly experimental and dirty) changes to a public branch?
Maybe someone gets inspired? (maybe me)

fraimondo, 2017-03-11:
Regarding the toolpath in the image, I expected to be able to cut a square hole in 0.5mm steps up to -5 (the -5/-4 location is just illustrative). The way I see it, it's cutting both sides of the path, and not going down up to -5mm like it happens when I use the same settings put no pocketing parameter.

Here you can see my [pycam fork](https://github.com/fraimondo/pycam)

I've also corrected a bug with toolpath cloning. Although there's still one more. I'll open a new issue here.

sumpfralle, 2017-03-11:
Interesting.
If the toolpath is moving along the right direction, but just fails to go down to the proper level, then it should be easy to fix. It would mean that just handling of the data exchange within the python module needs a fix.

Thank you for the toolpath cloning hint!
Feel free to create a pull request for these changes. This will retain your authorship of these commits.

palmerr23, 2017-06-12:
I, too, use the engrave offset parameter quite regularly in 0.5.1

My use is in creating wood inlays - positive / negative offsets for the solid piece and the pocket. 0.5.1 didn't work perfectly (you have to watch corner radii carefully) but it was a whole lot better than nothing!

For this work I quite often use tiny tools e.g. 0.8-1.5 mm mm router bits and travel very slowly.

I'd vote for its reinstatement as an explicit option. A default setting of tool diameter/2 would be fine. 

Richard

sumpfralle, 2017-06-15:
Thus your use-case requires a variety of different offset distances - not just the tool radius?
(just for clarification)

palmerr23, 2017-06-16:
Yes, I adjust the fit of the piece into the pocket by minute variations (0.1-0.2mm) of the offset on one or the other.

BTW. I'm very pleased to see development again on Pycam - it's my overall favourite CAM tool to use with LinuxCNC