SebKuzminsky/pycam #49: Crash on startup

Yinameah, 2017-04-02:
I'm trying to run PyCam with no success. I got : seg fault (core dump)

After digging some time, I found what caused it, or at least when It occurs. The faulty line is in  /scripts/pycam in line 147 :
`gui.reset_preferences()`

And I track it further. The problem comes from displaying the axes. If I put 
`"show_axes": False,` in line 24, /pycam/Gui/_ _init_ _.py the program launch correctly.

I don't if it's a bug in pycam or a configuration problem on my machine. I'm running Archlinux, up-to-date. PyCam v.0.5.1 installed from AUR work flawlessly.

I'm open to any suggestion...

sumpfralle, 2017-04-02:
This is interesting.
It sounds like this is related to some OpenGL function calls.

Below is the relevant code (see `pycam/Plugins/OpenGLViewAxes.py`):
```
        if self.core.get("view_light"):                                                             
            GL.glDisable(GL.GL_LIGHTING)                                                            
        GL.glLineWidth(1.5)                                                                         
        # draw a colored line ending in a cone for each axis                                        
        for index in range(3):                                                                      
            end = [0, 0, 0]                                                                         
            end[index] = length                                                                     
            color = [0.0, 0.0, 0.0]                                                                 
            # reduced brightness (not 1.0)                                                          
            color[index] = 0.8                                                                      
            GL.glColor3f(*color)                                                                    
            # we need to wait until the color change is active                                      
            GL.glFinish()                                                                           
            GL.glBegin(GL.GL_LINES)                                                                 
            GL.glVertex3f(*origin)                                                                  
            GL.glVertex3f(*end)                                                                     
            GL.glEnd()                                                                              
            # Position the cone slightly behind the end of the line - otherwise                     
            # the end of the line (width=2) is visible at the top of the cone.                      
            draw_direction_cone(origin, end, position=1.0 + cone_length, precision=32,              
                                size=cone_length)                                                   
        GL.glLineWidth(old_line_width)                                                              
        if self.core.get("view_light"):                                                             
            GL.glEnable(GL.GL_LIGHTING)                                                             
```

Could you try to disable ```view_light```? Does this prevent the error?

Or are you willing to experiment with the code snippet above?
(personally I would not put too much effort into it, since we will hopefully switch to a different visualization engine in the near future)

Yinameah, 2017-04-03:
Thank you for your answer.

With your informations, I tracked the problem further. The crash occurs in /pycam/Gui/OpenGLTools at the end of file (l. 85) :
`GLUT.glutSolidCone(cone_radius, cone_length, precision, 1)`

I managed to reproduce the crash in a terminal only importing GLUT and calling `GLUT.glutInit()` and then `GLUT.glutSolidCone(1, 1, 1, 1)`.

I've absolutely no background using GLUT neither OpenGL, so I can't tell if that's a glut problem or if something is wrongly initialized, or if some library is missing on my machine.

Does it make any sense to you ?

sumpfralle, 2017-04-03:
This is interesting!

I tried your minimal segfaulting setup here:
```
from OpenGL import GLUT
GLUT.glutInit()
GLUT.glutSolidCone(1, 1, 1, 1)
```
For me it does not fail.

I think, you should file a bug report against `python-opengl` in Archlinux.

Regarding a workaround in / for pycam: I am not fluent in OpenGL, thus I cannot think of an alternative for the above call. I think, we should add this error to the FAQ and explain how to work around it.
Could you post your error message and the diff that fixes the problem for you?

Yinameah, 2017-04-03:
Very well...
I reported the bug in Archlinux.

A simple work around is to comment the faulty line (l.85 in pycam/Gui/OpenGLTools.py)
The cones won't display, but I didn't noticed any other consequence.

The only error message I get is `Seg Fault (core dump)`. Altought I found more informations in system log. journalclt says :
`systemd-coredump[23481]: Process 23440 (python2) of user 1000 dumped core.`
                                                       
                                                       Stack trace of thread 23440:
                                                       #0  0x00007fbbd4e94323 fghDrawGeometrySolid (libglut.so.3)
                                                       #1  0x00007fbbd4e96d1a n/a (libglut.so.3)
                                                       #2  0x00007fbbd8b8e1c8 ffi_call_unix64 (libffi.so.6)
                                                       #3  0x00007fbbd8b8dc2a ffi_call (libffi.so.6)
                                                       #4  0x00007fbbd8da2927 _ctypes_callproc (_ctypes.so)
                                                       #5  0x00007fbbd8d9bd81 n/a (_ctypes.so)
                                                       #6  0x00007fbbda7a1692 PyObject_Call (libpython2.7.so.1.0)
                                                       #7  0x00007fbbda83f503 PyEval_EvalFrameEx (libpython2.7.so.1.0)
                                                       #8  0x00007fbbda8438dc PyEval_EvalCodeEx (libpython2.7.so.1.0)
                                                       #9  0x00007fbbda8439e8 PyEval_EvalCode (libpython2.7.so.1.0)
                                                       #10 0x00007fbbda85defe run_mod (libpython2.7.so.1.0)
                                                       #11 0x00007fbbda860139 PyRun_InteractiveOneFlags (libpython2.7.so.1.0)
                                                       #12 0x00007fbbda86032e PyRun_InteractiveLoopFlags (libpython2.7.so.1.0)
                                                       #13 0x00007fbbda8609dd PyRun_AnyFileExFlags (libpython2.7.so.1.0)
                                                       #14 0x00007fbbda872ca0 Py_Main (libpython2.7.so.1.0)
                                                       #15 0x00007fbbda1b3511 __libc_start_main (libc.so.6)
                                                       #16 0x000055bf6c4c17ea _start (python2.7)

Thanks for your time. I'll see if something goes out from bug report. I'll report it here in case.

Congratulation for your work with pycam by the way. We need software like that in the opensource community.

thawkins, 2017-04-10:
I have exactly the same problem on Fedora25. I will try the workarounds and see if that clears the problem.
Note: f25 runs under Wayland.

sumpfralle, 2017-04-29:
@thawkins: did you find the time to take a deeper look at fedora25?

l29ah, 2017-05-29:
Same for me on Gentoo amd64.

toms001, 2017-08-14:
If you expand the test to

from OpenGL import GLUT
GLUT.glutInit()
GLUT.glutCreateWindow('Testname')
GLUT.glutSolidCone(1, 1, 1, 1)

it should work with PyOpenGL - seems to be a init-problem.

I patched the Plugins/OpenGLWindow.py at line 567 to 
```
#       GL.glColorMaterial(GL.GL_FRONT_AND_BACK, GL.GL_EMISSION)
        GL.glEnable(GL.GL_COLOR_MATERIAL)
+       GLUT.glutCreateWindow('TemporaryCreateWindow')
```
an it works (with a new unnessary window) - maybe a hint for that problem, i don't know.

sumpfralle, 2017-08-15:
@toms001: thank you for taking a closer look.

Since we cannot really detect if the broken situation exists (it works fine here here on Debian stretch, i386), I would prefer to apply a workaround with a low annoyance factor.
Thus: could you close the window right afterwards (however this is done)?

Otherwise I will just add your workaround to the FAQ.

tkurbad, 2017-10-16:
Hi,

I'm on Gentoo amd64 and tried pycam 0.6.1. For me, @toms001 workaround did not work.
However, if I replace:
```
GLUT.glutSolidCone(...)
```
at the end of GUI/OpenGLTools.py with
```
GLUT.glutWireCone(...)
```
everything works (and the coordinate system has cones ;-)

Best,
Torsten

sumpfralle, 2017-10-16:
I applied your suggestion: 62694051b491bae314629446abbc58dbb3924baa

Thank you for tackling this weirdness!

closed 2017-10-16 by sumpfralle

SebKuzminsky, 2017-10-17:
Thanks @sumpfralle for applying this fix to 0.6 too.

Next time a commit should be in both 0.6 and in master, consider applying it in 0.6, then merging 0.6 into master (with `git checkout master`, `git merge origin/stable/0.6`).  That will make it more clear to future code-spelunkers that it really is the same commit, and that it went into both branches.

sumpfralle, 2017-10-17:
Ah - ok. Indeed I cherry-picked it, which creates a different line of history. Next time it will look more tree-like :)

commit e4cb8d42 (2017-10-16): avoid OpenGL issue with using "glutWireCone" instead of "glutSolidCone"

based on a hint from Torsten:
 https://github.com/SebKuzminsky/pycam/issues/49#issuecomment-336969101

Closes: #49
commit 62694051 (2017-10-16): avoid OpenGL issue with using "glutWireCone" instead of "glutSolidCone"

based on a hint from Torsten:
 https://github.com/SebKuzminsky/pycam/issues/49#issuecomment-336969101

Closes: #49