SebKuzminsky/pycam #51: PyCAM 0.6.1 on openSUSE leap 42.2 - error in visualisation, model not visible

cki61, 2017-04-17:
OS: openSUSE leap 42.2

Package Install:
zypper install python-gtk python-gtkglext python-opengl python-rsvg

Result:
python 2.7.12-24.1
python-gtk 2.24.0-16.6
python-gtkglext 1.1.0-5.1 (URL: http://download.opensuse.org/repositories/home:/bmwiedemann:/cnc/openSUSE_Leap_42.2/)
python-opengl 3.1.0-4.1
python-rsvg 2.32.0-1.1

Python Test:
test@opensuse422:~> python
Python 2.7.12 (default, Jun 28 2016, 06:57:42) [GCC] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gtk
>>> import OpenGL.GL
>>> import OpenGL.GLU
>>> import OpenGL.GLUT
>>> import gtk.gtkgl
>>> exit()
test@opensuse422:~> 

PyCAM Start:
test@opensuse422:~/D_IWS022/bin/pycam/scripts> ./pycam 
Enabled 4 parallel local processes
Traceback (most recent call last):
  File "/home/test/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 456, in check_busy_wrapper
    result = func(self, *args, **kwargs)
  File "/home/test/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 583, in gtkgl_functionwrapper_function
    self.glsetup()
  File "/home/test/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 486, in glsetup
    GLUT.glutInit()
  File "/usr/lib/python2.7/site-packages/OpenGL/GLUT/special.py", line 333, in glutInit
    _base_glutInit( ctypes.byref(count), holder )
  File "/usr/lib/python2.7/site-packages/OpenGL/platform/baseplatform.py", line 407, in __call__
    self.__name__, self.__name__,
OpenGL.error.NullFunctionError: Attempt to call an undefined function glutInit, check for bool(glutInit) before calling
Font directory: /home/test/D_IWS022/bin/pycam/share/fonts
Imported STL model: 0 vertices, 0 edges, 1444 triangles

PyCAM starts with errors seen above. A result seems to be that the model is not painted or invisible, please see appended image.
[image]

Has anybody an idea how to proceed?
Any proposals are welcome.

Reg.
CKI61

sumpfralle, 2017-04-17:
According to [this discussion](http://stackoverflow.com/questions/26700719/pyopengl-glutinit-nullfunctionerror) this could be a problem with your installation. Maybe the C library providing the GLUT functions is missing? A participant of the above thread suggests to install `freeglut3-dev` (package name in Debian). Maybe something similar would solve the problem for you?

Otherwise I would recommend to search for discussions about this problem related to openSUSE.

cki61, 2017-04-17:
Hi sumpfralle,
strike! I'd say.
I did a zypper install freeglut-devel (Version: 2.8.1-6.5) and the program is running.
Thank's a lot,

Best Regards,
cki61

closed 2017-04-17 by cki61

sumpfralle, 2017-04-17:
This is good to hear!
I added an error message referring to this package to the GLUT initialization: 352d1b457a67de56206ad105a9e6fea7eb0d8152