SebKuzminsky/pycam #52: Error while creating a new Process

valeriob01, 2017-04-18:
To reproduce
-----------------
- connect to a task server (running under pypy)
- try to create a new Process by clicking the New button

[image]

sumpfralle, 2017-04-18:
The top of the screenshot indicates that some plugins were not loaded. This is probably the reason for the missing "strategies", that cause the error.

Probably you should take a look at the first failing plugin and investigate, why it fails.

btw: I never heard of anyone using pypy with pycam. Thus you are entering new territory - have fun!

valeriob01, 2017-04-19:
On Tue, 2017-04-18 at 13:36 -0700, sumpfralle wrote:
> The top of the screenshot indicates that some plugins were not loaded. This is probably the reason for the missing "strategies", that cause the error.
> 
> Probably you should take a look at the first failing plugin and investigate, why it fails.

I have re-cloned pycam and started it normally, The first plugin to fail
is PathParameters.py, so probably you should also investigate.

> btw: I never heard of anyone using pypy with pycam. Thus you are entering new territory - have fun!

- The task server running under pypy, that was only a start at
experimentation.

valeriob01, 2017-04-19:
There are 3 plugins that fail, you should probably investigate also:
PathParameters, PathPatterns, and ProcessStrategies.
[image]

valeriob01, 2017-04-19:
Probably the last changes to the plugins in question "Switch POCKETING_TYPE_* to enum 'PocketingType'" is the problem...

valeriob01, 2017-04-19:
3 plugins are failing: PathParameters, PathPatterns and
ProcessStrategies,

The reason of the fail (I think): 

https://github.com/SebKuzminsky/pycam/commit/8a9bf8bf58fb748f51081aef8b8214b5bd637636

sumpfralle, 2017-04-19:
The above commit implies a new dependency on `enum34`. Did you install it?
I guess, this problem could be hidden by the plugin loader.

valeriob01, 2017-04-19:
No I didn't install it because nowhere a message about that.

valeriob01, 2017-04-20:
On Wed, 2017-04-19 at 11:07 -0700, sumpfralle wrote:
> The above commit implies a new dependency on `enum34`. Did you install it?
> I guess, this problem could be hidden by the plugin loader.

My attempt to install enum34 in Linuxcnc has failed, I switched to
another machine with Debian to see if I could install the python-enum34
package, obviously also wanted python-gtkglext1 and python-opengl.

I am going to close the issue.

closed 2017-04-20 by valeriob01