SebKuzminsky/pycam #54: Would be possible to change line to arc on reverse?

august9999, 2017-04-28:
Firstly many thanks for so grear job...
I'm using the software for abrasive waterjet machining toolpath generation. Surface processing with spheric tool is very similar to eroded profile.  In  AWJ machining controlled by traverse speed the jet erodes uninteruptedlly meanwile toolpath remains on the targe surface. On low speed I have deeper trench, on high speed I have shallow trench profiles. In G code, generated by pycam I change the z coordinate by feed speed value, according to this depth. So in my g code z coordinate is constant, the tool moves from start to finish point tracking zigzag toolpath changing the feed speeds instead changing z. In pycam the lines are jointed with parallel lines and on the angles and I have 0 speed for short time, but it is enouth to erode deeper areas there. So my question is:
Is it possible to change the line to the arc there? And if the answer is positive maybe could you direct me to the program part, where changes are necessaries?

sumpfralle, 2017-04-28:
What an interesting use-case! Thank you for sharing.

DId you already take a look at the corner precision? This is a GCode setting that controls the behaviour of the machines while changing direction. Take a look at the [G61 command](http://www.linuxcnc.org/docs/html/gcode/g-code.html#gcode:g61-g61.1) in the GCode documentation of [LinuxCNC](http://linuxcnc.org).

Could this solve your problem?

august9999, 2017-05-01:
...unfortunately NUM Flexium controller (

sumpfralle, 2017-05-01:
I assume that this controller does not support G61?

I am not really sure, how to handle this: it feels to me like a task for the machine controller ("maintain a high speed level").

If I would implement something similar (e.g. a toolpath filter for turning all changes of direction into an arc), I would probably re-implement the G61 feature of Linuxcnc. This would feel somehow wrong to me.

Is there no chance for you to switch to a better controller software?

sumpfralle, 2017-05-01:
Thinking about it further: this toolpath filter would probably be very trivial - thus maybe it is an option.

Technically it means the following:
* every switch of direction gets turned into an arc with a given "accuracy" (radius)
* a full turnabout (180 degree) cannot be turned into an arc - thus it would stay unchanged

Would this solve your problem?

august9999, 2017-05-02:
On this controller (num flexium) I have an option to set the maximal speed limit on the radius giving the error between the real and the programmed paths. By setting greater error value in controller allows to pass the programmed radius by higher speed. If the programmed speed exceeds limited speed it slows down to the limited  passing the radius by execution.

Your idea would solve this problem completely.