SebKuzminsky/pycam #55: ContourFollow Slow as Hell even for simple objects

Harvie, 2017-04-30:
Hello,
is there some reason why ContourFollow is so painfully slow even when using pypy? The basic idea seems quite simple to me... I've seen STL slicing be done really fast. And generating toolpaths from slice outlines also sounds like very easy and deterministic task that should not take more than few miliseconds... What exactly makes this so slow?

sumpfralle, 2017-04-30:
Currently it uses no external libraries, but rather imperfect prototyped code.
You are kindly invited to take a closer look and improve the situation. As you noticed: even a little effort will probably make a huge difference in that field.

Harvie, 2017-05-01:
Maybe PyCAM can reuse some code from blendercam. It's quite similar and is also written in python.
https://github.com/vilemnovak/blendercam

Or even better... You maybe you can even join your efforts together to create CAM library in python that would be used in both PyCAM and blendercam.

sumpfralle, 2017-05-01:
Agreed. I would be happy if someone would work on this.

Harvie, 2017-05-01:
Maybe opencamlib? but it's written in cpp.
https://github.com/aewallin/opencamlib

sumpfralle, 2017-05-01:
Yes, indeed - this would be great. Issue #25 outlines potential steps for a possible integration.

Harvie, 2017-05-01:
Oh, i see. blendercam is going for that too, so i think there's no point for this issue...

closed 2017-05-01 by Harvie