SebKuzminsky/pycam #56: gtk2 dependency problem

tcbetts, 2017-05-14:
pycam-0.6.1 reports gtk2 missing on ubuntu 16.04 installation - even after all dependencies have been loaded and verified.

sumpfralle, 2017-05-14:
Could you show us the list of installed python packages? E.g. `pip freeze` or `dpkg -l | grep python`.

tcbetts, 2017-05-14:
Here's the file...

ii  dh-python                                            2.20151103ubuntu1.1                           all          Debian helper tools for packaging Python libraries and applications
ii  drpython                                             1:3.11.4-1.1                                  all          simple and customizable editor for the Python language
ii  idle-python2.7                                       2.7.12-1ubuntu0~16.04.1                       all          IDE for Python (v2.7) using Tkinter
ii  ipython                                              2.4.1-1                                       all          enhanced interactive Python shell
ii  ipython-notebook                                     2.4.1-1                                       all          interactive Python html notebook
ii  ipython-notebook-common                              2.4.1-1                                       all          interactive Python html notebook data package
ii  libpeas-1.0-0-python3loader                          1.16.0-1ubuntu2                               amd64        Application plugin library
ii  libpython-all-dbg:amd64                              2.7.11-1                                      amd64        package depending on all supported Python debugging packages
ii  libpython-all-dev:amd64                              2.7.11-1                                      amd64        package depending on all supported Python development packages
ii  libpython-dbg:amd64                                  2.7.11-1                                      amd64        debug build of the Python Interpreter (version 2.7)
ii  libpython-dev:amd64                                  2.7.11-1                                      amd64        header files and a static library for Python (default)
ii  libpython-stdlib:amd64                               2.7.11-1                                      amd64        interactive high-level object-oriented language (default python version)
ii  libpython2.7:amd64                                   2.7.12-1ubuntu0~16.04.1                       amd64        Shared Python runtime library (version 2.7)
ii  libpython2.7-dbg:amd64                               2.7.12-1ubuntu0~16.04.1                       amd64        Debug Build of the Python Interpreter (version 2.7)
ii  libpython2.7-dev:amd64                               2.7.12-1ubuntu0~16.04.1                       amd64        Header files and a static library for Python (v2.7)
ii  libpython2.7-minimal:amd64                           2.7.12-1ubuntu0~16.04.1                       amd64        Minimal subset of the Python language (version 2.7)
ii  libpython2.7-stdlib:amd64                            2.7.12-1ubuntu0~16.04.1                       amd64        Interactive high-level object-oriented language (standard library, version 2.7)
ii  libpython3-all-dbg:amd64                             3.5.1-3                                       amd64        package depending on all supported Python 3 debugging packages
ii  libpython3-all-dev:amd64                             3.5.1-3                                       amd64        package depending on all supported Python 3 development packages
ii  libpython3-dbg:amd64                                 3.5.1-3                                       amd64        debug build of the Python 3 Interpreter (version 3.5)
ii  libpython3-dev:amd64                                 3.5.1-3                                       amd64        header files and a static library for Python (default)
ii  libpython3-stdlib:amd64                              3.5.1-3                                       amd64        interactive high-level object-oriented language (default python3 version)
rc  libpython3.4:amd64                                   3.4.3-1ubuntu1~14.04.5                        amd64        Shared Python runtime library (version 3.4)
rc  libpython3.4-minimal:amd64                           3.4.3-1ubuntu1~14.04.5                        amd64        Minimal subset of the Python language (version 3.4)
ii  libpython3.5:amd64                                   3.5.2-2ubuntu0~16.04.1                        amd64        Shared Python runtime library (version 3.5)
ii  libpython3.5-dbg:amd64                               3.5.2-2ubuntu0~16.04.1                        amd64        Debug Build of the Python Interpreter (version 3.5)
ii  libpython3.5-dev:amd64                               3.5.2-2ubuntu0~16.04.1                        amd64        Header files and a static library for Python (v3.5)
ii  libpython3.5-minimal:amd64                           3.5.2-2ubuntu0~16.04.1                        amd64        Minimal subset of the Python language (version 3.5)
ii  libpython3.5-stdlib:amd64                            3.5.2-2ubuntu0~16.04.1                        amd64        Interactive high-level object-oriented language (standard library, version 3.5)
ii  python                                               2.7.11-1                                      amd64        interactive high-level object-oriented language (default version)
ii  python-alabaster                                     0.7.7-1                                       all          Configurable sidebar-enabled Sphinx theme (Python 2)
ii  python-all                                           2.7.11-1                                      amd64        package depending on all supported Python runtime versions
[... cut ...]
ii  python3.5-dbg                                        3.5.2-2ubuntu0~16.04.1                        amd64        Debug Build of the Python Interpreter (version 3.5)
ii  python3.5-dev                                        3.5.2-2ubuntu0~16.04.1                        amd64        Header files and a static library for Python (v3.5)
ii  python3.5-minimal                                    3.5.2-2ubuntu0~16.04.1                        amd64        Minimal subset of the Python language (version 3.5)

sumpfralle, 2017-05-14:
Interesting.
What happens if you start `python` and `import gtk`?
Do you have other ideas, why GTK could fail on your system?

tcbetts, 2017-05-14:
Invoking python at the terminal loads Anaconda python 2.7. (along with a
slew of messages). Entering import numpy as np (as a check) generates more
messages, but works. Entering import gkt or gkt2 yields the message:
"ImportError: No module named gkt", etc.

Looks like my python has been hijacked!

tcbetts, 2017-05-14:
Can I install pycam from within Anaconda's python?

tcbetts, 2017-05-14:
I think I've solved the problem... First, I commented out the Anaconda PATH
in .bashrc using vim. Then, I executed ./pycam in the script folder within
my pycam-0.6.1 directory.  This brought up PYCAM and PyCam Visualization
windows which seem to be working.

Only problem, no help file. I want to load STL files and generate G-Code
output. Where can I find instructions?

sumpfralle, 2017-05-14:
Interesting!
It seems like "anaconda" preconfigures some kind of "virtualenv" for you - and hides packages, that are installed system-wide. Maybe `toggleglobalsitepackages` (switch to `enabled`) would also solve your problem without disabling "anaconda" completely? Probably the anaconda documentation contains more information regarding the usage of external programs with python dependencies.

You can find pycam's documentation directly on its website: http://pycam.sf.net/

laconc, 2017-06-28:
The original issue has been resolved and @tcbetts was directed to the documentation that he was seeking. Appears like this issue can be closed.

sumpfralle, 2017-06-28:
Correct - finally the issue count is going down by itself :)

closed 2017-06-28 by sumpfralle