SebKuzminsky/pycam #57: Surfacing Causes Exception

1-23, 2017-05-17:
Text of exception is:

An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot!
Traceback (most recent call last):
  File "/home/jacob/Downloads/pycam-0.6.1/pycam/Plugins/Tasks.py", line 318, in generate_toolpath
    toolpath = task_resolver(task, callback=draw_callback)
  File "/home/jacob/Downloads/pycam-0.6.1/pycam/Plugins/TaskTypes.py", line 70, in run_task
    maxz=box.upper.z, draw_callback=callback)
  File "/home/jacob/Downloads/pycam-0.6.1/pycam/PathGenerators/DropCutter.py", line 69, in GenerateToolPath
    callback=progress_counter.update):
  File "/home/jacob/Downloads/pycam-0.6.1/pycam/Utils/threading.py", line 649, in run_in_parallel_local
    for result in imap_func(func, args):
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 668, in next
    raise value
TypeError: 'NoneType' object has no attribute '__getitem__'

sumpfralle, 2017-05-20:
Does the surfacing operation work without parallel processing? (`Settings -> Parallel Processing`)

1-23, 2017-05-23:
Yes it does.

dirkvs, 2017-06-09:
I'm getting exactly the same issue, but it doesn't work with parallel processing turned off.

dirkvs, 2017-06-09:
Here's some more information. It looks like somehow multiprocessing is re-enabled when generating the toolpath, or not disabled properly - see below. The error then re-occurs.

If I disable multiprocessing then quit and restart, the program restarts with multiprocessing enabled again.

11:00 INFO Enabled 8 parallel local processes
11:00 INFO Font directory: /home/dirk/pyCAM/pycam-0.6.1/share/fonts
11:00 INFO Imported STL model: 0 vertices, 0 edges, 1444 triangles
11:01 INFO Imported STL model: 0 vertices, 0 edges, 64 triangles
11:04 ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/Tasks.py", line 318, in generate_toolpath     toolpath = task_resolver(task, callback=draw_callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/TaskTypes.py", line 70, in run_task     maxz=box.upper.z, draw_callback=callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/PathGenerators/DropCutter.py", line 69, in GenerateToolPath     callback=progress_counter.update):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Utils/threading.py", line 649, in run_in_parallel_local     for result in imap_func(func, args):   File "/usr/lib/python2.7/multiprocessing/pool.py", line 668, in next     raise value TypeError: 'NoneType' object has no attribute '__getitem__'
**11:05 INFO Multiprocessing disabled
11:05 INFO Enabled 8 parallel local processes**
11:05 ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/Tasks.py", line 318, in generate_toolpath     toolpath = task_resolver(task, callback=draw_callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/TaskTypes.py", line 70, in run_task     maxz=box.upper.z, draw_callback=callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/PathGenerators/DropCutter.py", line 69, in GenerateToolPath     callback=progress_counter.update):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Utils/threading.py", line 649, in run_in_parallel_local     for result in imap_func(func, args):   File "/usr/lib/python2.7/multiprocessing/pool.py", line 668, in next     raise value TypeError: 'NoneType' object has no attribute '__getitem__'
11:06 ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/Tasks.py", line 318, in generate_toolpath     toolpath = task_resolver(task, callback=draw_callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/TaskTypes.py", line 70, in run_task     maxz=box.upper.z, draw_callback=callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/PathGenerators/DropCutter.py", line 69, in GenerateToolPath     callback=progress_counter.update):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Utils/threading.py", line 649, in run_in_parallel_local     for result in imap_func(func, args):   File "/usr/lib/python2.7/multiprocessing/pool.py", line 668, in next     raise value TypeError: 'NoneType' object has no attribute '__getitem__'
11:06 ERROR An unexpected exception occoured: please send the text below to the developers of PyCAM. Thanks a lot! Traceback (most recent call last):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/Tasks.py", line 318, in generate_toolpath     toolpath = task_resolver(task, callback=draw_callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Plugins/TaskTypes.py", line 70, in run_task     maxz=box.upper.z, draw_callback=callback)   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/PathGenerators/DropCutter.py", line 69, in GenerateToolPath     callback=progress_counter.update):   File "/home/dirk/pyCAM/pycam-0.6.1/pycam/Utils/threading.py", line 649, in run_in_parallel_local     for result in imap_func(func, args):   File "/usr/lib/python2.7/multiprocessing/pool.py", line 668, in next     raise value TypeError: 'NoneType' object has no attribute '__getitem__'