SebKuzminsky/pycam #58: Add support for step models?

TurBoss, 2017-05-20:
Hello

it would be very cool to have support to open step models

this can be done with OCE 
its only a suggestion

Thanks

sumpfralle, 2017-05-21:
Another import format would surely be a good thing.

Anyone interested in this feature may want to take a look at `pycam/Importers/STLImporter.py`. Basically you only need to create a function that receives a file-like handle and returns a mesh model (made of triangles). Afterwards you can add this new importer function to the list of handled import formats in `pycam/Importers/__init__.py`. Now it should be available everywhere in pycam.

Pull requests are welcome!

TurBoss, 2017-05-21:
Investigating pythonocc-core to see if I can do something...
I can't promise anything :D

TurBoss, 2017-05-26:
well did a bit of research and found that I'm not able to accomplish this task :D

sumpfralle, 2017-05-26:
What a pity!

Out of curiosity: is the step format too complicated  or incompatible with a triangle mesh based data structure? Or do you think that you currently lack the python skills?
(I am not trying to persuade you - I just want to prepare a good starting point for people coming back to this topic later on)

TurBoss, 2017-05-26:
Hi

its not triangle oriented example:
a curve is not composed of a lot of vertices is just a curve 

I'm lacking python and opencascade skills

what I have looked at is:
http://www.pythonocc.org/

will spend some time on this :D

TurBoss, 2017-05-26:
here is an example of a step model: http://www.pythonocc.org/quick-examples/step-ap203-import/

sumpfralle, 2017-05-26:
Maybe pythonocc can export a model to STL? This would be the most straight-forward approach for importing step files into pycam.

TurBoss, 2017-05-26:
stl still a mesh based format not a cad model

sumpfralle, 2017-05-27:
> stl still a mesh based format not a cad model

Yes, indeed. Pycam uses triangle meshs internally for 3D models. Thus this no problem.
I assumed that you were contemplating toolpath generation based on 3D models loaded from step files, right? Or did have something else in mind?

> I can export in stl from freecad

If freecad supports a non-interactive conversion from step to stl, then this could be a simple option, too. You could take a look at pycam's `PSImporter` - it uses inkscape for converting (e)ps files to DXF (which can be parsed by pycam).

palmerr23, 2017-06-12:
Moving between the two CAD modeling schemes IGES and STEP is a non-trivial issue and has been discussed extensively in CAD forums and there are several (at least partly) successful translation options, most using CAD software that reads/writes both types of model.

I've found that FreeCAD does the trick... most of the time.