SebKuzminsky/pycam #59: latest commit

valeriob01, 2017-05-27:
~/test/pycam/scripts$ ./pycam
Enabled 4 parallel local processes
Traceback (most recent call last):
  File "./pycam", line 725, in <module>
    exit_code = execute(parser, opts, args, pycam)
  File "./pycam", line 305, in execute
    result = show_gui(inputfile, opts.config_file)
  File "./pycam", line 142, in show_gui
    gui = gui_class(event_manager)
  File "/home/sel/test/pycam/pycam/Gui/Project.py", line 184, in __init__
    self.gui.get_object("AboutWindowButtons").get_children()[-1].connect(
IndexError: list index out of range

sumpfralle, 2017-05-30:
Thank you!

Could you try this again, please with the current master branch?

Please also try it with python3 (`python3 scripts/pycam`). Python2-compatibility will go away soon, anyway.

valeriob01, 2017-05-30:
Same error, with and without python3

~/test/pycam/scripts$ python3 ./pycam
Enabled 4 parallel local processes
Traceback (most recent call last):
  File "./pycam", line 725, in <module>
    exit_code = execute(parser, opts, args, pycam)
  File "./pycam", line 305, in execute
    result = show_gui(inputfile, opts.config_file)
  File "./pycam", line 142, in show_gui
    gui = gui_class(event_manager)
  File "/home/sel/test/pycam/pycam/Gui/Project.py", line 184, in __init__
    self.gui.get_object("AboutWindowButtons").get_children()[-1].connect(
IndexError: list index out of range

sumpfralle, 2017-05-30:
Very weird.

Could you add `print(list(self.gui.get_object("AboutWindowButtons").get_children()))` before line 184 in pycam/Gui/Project.py? Do you see additional output?

valeriob01, 2017-05-30:
I am not a python programmer but that [-1] there is suspect, if it is a list index, it should be out of range...

ebo, 2017-05-30:
On May 30 2017 6:12 AM, valeriob01 wrote:
> I am not a python programmer but that [-1] there is suspect, if it is
> a list index, it should be out of range...

Without a code snittpett it is hard to geuss exactly.  That said, I am 
going to guess that it is a standard index from the end of an array or 
list.

 From the NumPy docs:

"Negative i and j are interpreted as n + i and n + j where n is the 
number of elements in the corresponding dimension. Negative k makes 
stepping go towards smaller indices.

"Example:
>>> x = np.array([0, 1, 2, 3, 4, 5, 6, 7, 8, 9])
...
>>> x[-2:10]
array([8, 9])
"

Hope this helps.

   EBo --

valeriob01, 2017-05-30:
I have added the mentioned line, which prints nothing:

        # "about" window
        self.about_window = self.gui.get_object("AboutWindow")
        self.about_window.set_version(VERSION)
        self.gui.get_object("About").connect("activate", self.toggle_about_window, True)
        # we assume, that the last child of the window is the "close" button
        # TODO: fix this ugly hack!
	print(list(self.gui.get_object("AboutWindowButtons").get_children()))
	self.gui.get_object("AboutWindowButtons").get_children()[-1].connect(
            "clicked", self.toggle_about_window, False)
        self.about_window.connect("delete-event", self.toggle_about_window, False)

valeriob01, 2017-05-30:
~/test/pycam/scripts$ ./pycam
Enabled 4 parallel local processes
[]
Traceback (most recent call last):
  File "./pycam", line 725, in <module>
    exit_code = execute(parser, opts, args, pycam)
  File "./pycam", line 305, in execute
    result = show_gui(inputfile, opts.config_file)
  File "./pycam", line 142, in show_gui
    gui = gui_class(event_manager)
  File "/home/sel/test/pycam/pycam/Gui/Project.py", line 185, in __init__
    self.gui.get_object("AboutWindowButtons").get_children()[-1].connect(
IndexError: list index out of range

ebo, 2017-05-30:
On May 30 2017 8:20 AM, valeriob01 wrote:
> ~/test/pycam/scripts$ ./pycam
> Enabled 4 parallel local processes
> []
> Traceback (most recent call last):
>   File "./pycam", line 725, in <module>
>     exit_code = execute(parser, opts, args, pycam) [...]

try printing/reporting the results of 
self.gui.get_object("AboutWindowButtons").get_children()  You may find 
that it returns [] or None.  If it is anything else I will be a bit 
surprised.

ebo, 2017-05-30:
On May 30 2017 8:18 AM, valeriob01 wrote:
> I have added the mentioned line, which prints nothing:
>
>         # "about" window
>         self.about_window = self.gui.get_object("AboutWindow")
>         self.about_window.set_version(VERSION) [...]

sorry.  I sent the previous email before reading this one.  You 
anticipated me.

I would modify the following:

print("children=%s"%srt(list(self.gui.get_object("AboutWindowButtons").get_children())))

You might have to jiggle the above a little to get it to work depending 
on type, BUT what I expect is happening is that there are no children.  
If so, then we should define what should happen --a and I am guessing 
that we should just ignore it.

   EBo --

valeriob01, 2017-05-30:
Well, I had not noticed that it printed [], so I assume that the list is empty

ebo, 2017-05-30:
On May 30 2017 8:29 AM, valeriob01 wrote:
> Well, I had not noticed that it printed [], so I assume that the list 
> is empty

that would be my guess.

Is it valid to be empty and be called?  If so, then I would just return 
or branch around.  Otherwise there appears to be a bug.

Anyway, that would be my guess.

   EBo --

sumpfralle, 2017-05-30:
Regarding the python-side of the things:
* `foo[-1]` is a valid expression for "the last item of a list"
* `foo()[n]` causes an `IndexError` if `foo` is empty (this is happening here)

I am slightly confused, that your (@valeriob01) "about" window has no children. Weird.

Anyway: this situation is now handled more carefully.
Please report back:
* if pycam starts for you now
* if you are able to close the "about" window

closed 2017-05-30 by sumpfralle

valeriob01, 2017-05-31:
Reading the code snippet again, it seems that the About Window has no buttons. This is possible, to close the window just click the X at top-right of window.

valeriob01, 2017-05-31:
Lars, I have tested the master, it starts with an error and the 3D view is not working. 
[image]

Regarding the About window: it has 2 buttons. Regarding the other windows and menus, there are serious bugs:
[image]

sumpfralle, 2017-05-31:
Thank you for testing this.

The error regarding the 3D view just reflects the current situation, that someone will need to work on that, since this feature is just missing at the moment.

Regarding other GTK problems: I hope, that other people who care about pycam's GUI will start to work on these issues. Thus if you feel like reporting and documenting such glitches, then I would recommend to collect them all in a single issue ticket.

commit 0291a501 (2017-05-30): tolerate empty children list of "about" window

Closes: #59