SebKuzminsky/pycam #60: Specify multiple servers when distributed processing mode is enabled

valeriob01, 2017-06-21:
This enhancement could bring the possibility to specify a list of servers, and each server is used in parallel with other servers to increase the number of available cores.

sumpfralle, 2017-06-24:
Personally I am not in favor of putting more effort into the distributed processing. It is is a nice feature, but it is also probably the third most often mentioned cause of problem reports (besides the OpenGL visualization and toolpath algorithms). Thus I am not confident, that the effort put into this feature is justified.
Anyway: patches and pull requests are welcome!

valeriob01, 2019-08-27:
This could introduce more bugs than features. It works well now as it is. With the number of cores available at good price it is not an issue anymore.

closed 2019-08-27 by valeriob01