SebKuzminsky/pycam #61: Docker image

laconc, 2017-06-22:
I made a quick docker image to get pycam on my mac. Is there any interest in the idea of me polishing it and creating a PR?

[image]

matelakat, 2017-06-22:
@laconc yes please, I am interested in this.

ebo, 2017-06-22:
Oh YEA...  This will allow people to set up an alternative install 
method, so that if they have issues with a regular source install, they 
can get something "just working".  Also, it would allow for 
configuration and compatibility testing (think test driven development, 
etc.).  Good stuff.

On Jun 21 2017 9:57 PM, Dashiel Lopez Mendez wrote:
> I made a quick docker image to get pycam on my mac. Is there any
> interest in the idea of me polishing it and creating a PR?
>
> ![screen shot 2017-06-21 at 10 45 41
> 
> pm](https://user-images.githubusercontent.com/20423536/27416606-1969c850-56d4-11e7-8a89-60807d0ac342.png)

laconc, 2017-06-22:
Sweet, I'll get on it. I'm in the middle of finishing up some summer classes so I probably won't have something until next week.

matelakat, 2017-06-22:
@laconc could you just open a PR with what you have, mark it as WIP ?

laconc, 2017-06-22:
@matelakat done.

#62

laconc, 2017-06-28:
The PR has had quite a bit of development work done to it, closing this.

closed 2017-06-28 by laconc