SebKuzminsky/pycam #64: List index out of range

valeriob01, 2017-06-30:
When creating a new process:

[image]

sumpfralle, 2017-11-11:
Does this happen every time you start pycam? If yes: please show us the complete log.

Currently I cannot reproduce this.

valeriob01, 2017-11-11:
On Sat, 2017-11-11 at 03:59 +0000, sumpfralle wrote:
> Does this happen every time you start pycam? If yes: please show us the complete log.
> 
> Currently I cannot reproduce this.

This bug was related to an old version of pycam.

No surprise you cannot reproduce it now.

valeriob01, 2017-11-11:
On Sat, 2017-11-11 at 03:59 +0000, sumpfralle wrote:
> Does this happen every time you start pycam? If yes: please show us the complete log.
> 
> Currently I cannot reproduce this.

BTW, the screenshot shows the portion of visible output, and the trace
shows that various actions where made before triggering the bug

It does not show when you load pycam, it shows after various actions of
the user

sumpfralle, 2017-11-11:
> This bug was related to an old version of pycam.

In this case I suggest to leave it rest in peace :)

closed 2017-11-11 by sumpfralle