SebKuzminsky/pycam #65: Python 3: "ModuleNotFoundError: No moduled named 'urlparse'"

acer9997, 2017-07-04:
Trying to run with Python 3 on MacOS Sierra (10.12.5) and PyCam v0.6.1

```
➜  pycam-0.6.1 python3 ./scripts/pycam
Traceback (most recent call last):
  File "./scripts/pycam", line 54, in <module>
    from pycam.Geometry import Box3D, Point3D
  File "/Users/acer9997/Downloads/pycam-0.6.1/pycam/Geometry/__init__.py", line 26, in <module>
    import pycam.Utils.log
  File "/Users/acer9997/Downloads/pycam-0.6.1/pycam/Utils/__init__.py", line 27, in <module>
    import urlparse
ModuleNotFoundError: No module named 'urlparse'
```

laconc, 2017-07-04:
`urlparse` is one of the libraries that was restructured in the migration to Python 3, hence, that version of PyCam is only compatible with Python 2. You can try it with 2 but I suspect that it would still fail to run.

Alternatively, you could clone or download the latest code from here and follow the instructions in INSTALL.md. MacOS will still require that you use Python 2 though, due to the dependencies.

SebKuzminsky, 2017-10-11:
Can you check if this was fixed in commit  0cbd8b2596a9867c3e058198833c5d5d8be2dd2c, perhaps by testing the stable/0.6 branch?

sumpfralle, 2017-11-11:
I went through the log `pycam/Utils/__ini__.py`. I am confident, that the above issue was fixed in 1e1af2bc2887d3e0c79435d1e1eb0eedb22976fe and 0cbd8b2596a9867c3e058198833c5d5d8be2dd2c.

closed 2017-11-11 by sumpfralle