SebKuzminsky/pycam #67: Inconsistent strategy parameters input UI

l29ah, 2017-08-28:
https://dump.bitcheese.net/files/zulekev/pycam-inconsistent.ogv

sumpfralle, 2017-08-28:
Thank you for your report and the screen recording.

Other people encountering the same issue would have probably appreciated some words describing the situation. Videos are currently hard to index for search engines :(

How to reproduce:
* choose the processing strategy "slice removal"
* switch to another processing strategy
* switch back to "slice removal"
* seemingly randomly the parameters "direction" and "milling style" are both visible or both hidden when "slice removal" is selected

Expected behaviour:
* the parameters "direction" and "milling style" should always be visible after switching to the "slice removal" strategy

Or did I misunderstand a part of your bug report?

l29ah, 2017-08-28:
Yes, it's correct, thank you. I had a hard time finding the proper words for the report.

sumpfralle, 2017-08-29:
No problem.

Regarding the inconsistent behaviour: the model behind the GUI is changing at the moment. I will keep this bug report open as a reminder until that is finished.

sumpfralle, 2017-12-25:
I cannot reproduce this with the master branch anymore. Thus it is probably fixed.

closed 2017-12-25 by sumpfralle