SebKuzminsky/pycam #69: 'gi.repository.Gtk' object has no attribute 'gdkgl'

l29ah, 2017-08-28:
```
‰ pycam
fatal: Not a git repository (or any of the parent directories): .git
/usr/lib64/python2.7/site-packages/pycam/Plugins/__init__.py:27: PyGIWarning: Gtk was imported without specifying a version first. Use gi.require_version('Gtk', '3.0') before import to ensure that the right version gets loaded.
  from gi.repository import Gtk as gtk
Enabled 4 parallel local processes

(pycam:11485): Gtk-WARNING **: Theme directory 256x256@2/animations of theme hicolor has no size field

(pycam:11485): Gtk-CRITICAL **: gtk_grid_attach: assertion 'width > 0' failed

(pycam:11485): Gtk-CRITICAL **: gtk_grid_attach: assertion 'width > 0' failed

(pycam:11485): Gtk-CRITICAL **: gtk_grid_attach: assertion 'width > 0' failed

(pycam:11485): Gtk-CRITICAL **: gtk_grid_attach: assertion 'width > 0' failed
Traceback (most recent call last):
  File "/usr/lib/python-exec/python2.7/pycam", line 725, in <module>
    exit_code = execute(parser, opts, args, pycam)
  File "/usr/lib/python-exec/python2.7/pycam", line 305, in execute
    result = show_gui(inputfile, opts.config_file)
  File "/usr/lib/python-exec/python2.7/pycam", line 145, in show_gui
    plugin_manager.import_plugins()
  File "/usr/lib64/python2.7/site-packages/pycam/Plugins/__init__.py", line 258, in import_plugins
    self._load_plugin(plugin, filename, name)
  File "/usr/lib64/python2.7/site-packages/pycam/Plugins/__init__.py", line 280, in _load_plugin
    if not new_plugin.setup():
  File "/usr/lib64/python2.7/site-packages/pycam/Plugins/OpenGLWindow.py", line 174, in setup
    glconfig = self._gtk.gdkgl.Config(mode=(self._gtk.gdkgl.MODE_RGBA
  File "/usr/lib64/python2.7/site-packages/gi/overrides/__init__.py", line 39, in __getattr__
    return getattr(self._introspection_module, name)
  File "/usr/lib64/python2.7/site-packages/gi/module.py", line 139, in __getattr__
    self.__name__, name))
AttributeError: 'gi.repository.Gtk' object has no attribute 'gdkgl'
```

sumpfralle, 2017-09-03:
This is an (incomprehensible) way of telling you, that pycam does currently not support the opengl visualisation (since it moved to gtk3).

I just fixed this now by providing a more suitable warning message (see b0d0ecd815bd94e67d61e911493937dd7c2632c5).

closed 2017-09-03 by sumpfralle

l29ah, 2017-09-03:
How do i run it then?

sumpfralle, 2017-09-03:
The commit linked above (b0d0ecd815bd94e67d61e911493937dd7c2632c5) shows you how to modify your local pycam code in-place.

Afterwards pycam will mostly work - except for the 3D preview. Thus you will probably need to use something like the [linuxcnc simulator](http://linuxcnc.org) to preview your toolpath :(