SebKuzminsky/pycam #70: install does not work (at all) on Ubuntu 14.04

terramars, 2017-08-29:
when I run in command line mode, I get this error.

~/cnc/pycam$ scripts/pycam --export-gcode=test.ngc test.STEP 
Enabled 8 parallel local processes
Traceback (most recent call last):
  File "scripts/pycam", line 725, in <module>
    exit_code = execute(parser, opts, args, pycam)
  File "scripts/pycam", line 311, in execute
    tps = pycam.Gui.Settings.ToolpathSettings()
AttributeError: 'module' object has no attribute 'ToolpathSettings'

I also can't run in GUI mode but that's because I have an openGL problem.  Your installation and testing documentation is really lacking, I have no idea how to fix any of these issues despite following all the instructions.  When I try to install with setup.py, I just get this error 

Traceback (most recent call last):
  File "./build/scripts-2.7/pycam", line 53, in <module>
    import pycam.Exporters.GCodeExporter
ImportError: No module named Exporters.GCodeExporter

How do I get this running on 14.04?  GUI would be nice but I'm fine with command line.  Or am I fucked and need to upgrade to 16.04?

terramars, 2017-08-29:
after fixing the no module issue, i just get the gtk segfault error from mixing "import gtk" with the gobject way of importing gtk (in Gui/common.py)

~/cnc/pycam$ pycam
Enabled 8 parallel local processes
/usr/lib/python2.7/dist-packages/gobject/constants.py:24: Warning: g_boxed_type_register_static: assertion 'g_type_from_name (name) == 0' failed
  import gobject._gobject
/usr/lib/python2.7/dist-packages/gtk-2.0/gtk/__init__.py:40: Warning: specified class size for type 'PyGtkGenericCellRenderer' is smaller than the parent type's 'GtkCellRenderer' class size
  from gtk import _gtk
/usr/lib/python2.7/dist-packages/gtk-2.0/gtk/__init__.py:40: Warning: g_type_get_qdata: assertion 'node != NULL' failed
  from gtk import _gtk
Segmentation fault (core dumped)

I've tried fixing that and I just get version errors requiring 3.12 instead of 3.10...

sumpfralle, 2017-08-29:
Indeed the old commandline-based interface is not supported since v0.6 anymore. Sadly the `--help` switch of pycam still pretends that it is there :(

If the commandline is fine for you, then I would recommend to try the new configuration/script based workflow of pycam.

Take a look at the branch "yaml-flow" - the file "yaml_flow_working.yml" contains the current set of available features.

terramars, 2017-08-29:
oh great thanks i'll check it out.

i would like to get the UI version running on my ubuntu install, OpenGL
does work on my machine so that's not the only problem.  any tips?

sumpfralle, 2017-08-29:
If you want to use a reliably working GUI, then I would recommend pycam v0.5.1. It is a bit old, but does not lack any features. Additionally it should work well with your (slightly aged) system.

terramars, 2017-08-29:
ok great i'll try that

terramars, 2017-08-29:
just installed the 0.5.1 deb file on sourceforge - this is what I get :

~/cnc$ pycam
Traceback (most recent call last):
  File "/usr/bin/pycam", line 39, in <module>
    import pycam.Gui.Console
ImportError: No module named Console

terramars, 2017-08-29:
works using install from source!

SebKuzminsky, 2017-10-11:
This is a duplicate of #68.

closed 2017-10-11 by SebKuzminsky