SebKuzminsky/pycam #71: change the 10000rpm spindle speed limitation

lfc7, 2017-09-02:
Hi,
the spindle speed is limited to 10000rpm (if trying to set it faster, it remain at 10000), is there a way to change this to 20000rpm.
thanks

--
oliv

sumpfralle, 2017-09-02:
The GTK GUI defines this limit. Please take a look at the file `share/ui/tasks.ui` and change the maximum value for `speed` (defined in the last seven lines of that file).

What would be a reasonable (future-proof) maximum value for this setting from your point of view?

diggit, 2017-09-02:
Hi, CNC mill build I participate on has "china generic" spindle which can handle up to 18k rpm. There are spindles up to 60k rpm on ebay. 100k should be enough for future I think...

lfc7, 2017-09-02:
Yes got it, share/ui/tasks.ui is the right place but in the 7 **first** lines, it works fine, many thanks sumpfralle.
I agree with diggit, 100k seams to be enought.

closed 2017-09-02 by lfc7