SebKuzminsky/pycam #73: Plunge feedrate

august9999, 2017-09-15:
Hi Guys,
I'm using your code for 3D Abrasive Waterjet machining, it works perfect, it helps a lot!!! Many Thanks!!! but always I have to change the pycam generated g-code with regexes. On the right points I have to switch the abrasive on before the tool moves up and switch on again after tool plunges down. Of course it would much more comfortable if pycam would generate the final code. So I'm trying to find the way. I changed LinuxCNC.py to write the g-code as I need, but I faced with the problem:
In the g-code ONLY after "plunge feedrate" I have to input M..., but if I input in the LinuxCNC.py where the letter "F" is assigned  to  the "set feedrate",  it also appears and after  "feedrate" too. Maybe any idea how to solve it? thank you in advance,

sumpfralle, 2017-09-16:
Interesting!

Do you use M3/M5 for turning on and off the abrasive?

It feels to me, like you would just need a switch for "disable spindle during safety moves"?
Do I understand you correctly?

Can you think of a more generic term instead of "spindle", that would also cover your abrasive?

august9999, 2017-09-18:
Yes, you understood me correctly. I'd need a switch for "disable spindle during safety moves".

I'm using the switch M3/M5.

It switches on "M3" before the program starts and  switches off "M5" in the end of the program, but it not disables spindle during safety moves during the program.

Maybe I'm using it wrong?

Design: half sphere (like a igloo)
Pycam set up...
Strategy: slice removal
Pattern: Grid
Milling style: ignore

sumpfralle, 2017-09-18:
> Maybe I'm using it wrong?

You are doing everything correctly - the feature is just not there, yet.

Now I understand your goal. It looks like an acceptable challenge which is worth pursuing.
Stay tuned ...

august9999, 2017-09-18:
Thank you for your support...

august9999, 2017-11-13:
I applied few filters in the program, mainly in ToolpathExport.py using regexes and now it generates g code  switching on and off the spindle (of course it's not the best way, but anyway the goal is achieved).

I would like to ask you if would be possible to run the source code on windows?

Our microwaterjet machine is controlled by Beckhoff controller, Twin CAT 3 software with visual studio integration on windows.

august9999, 2017-11-13:
example of generated first layer of the sphere
[output_sph_EB_1x.txt](https://github.com/SebKuzminsky/pycam/files/1467135/output_sph_EB_1x.txt)

august9999, 2019-11-20:
problem solved

closed 2019-11-20 by august9999

sumpfralle, 2019-11-20:
> problem solved

I guess, you solved it only locally, right?
Indeed I consider this a useful feature, that is worth to be implemented. Thus I would re-open the issue. Or what do you think=

august9999, 2019-11-21:
Yes, I did it locally on 0.6.1 version, but the method itself, switching the jet on and off did not prove my expectations experimentally, so I quit it.  I had to change the code in GCodeExporter.py; ToolPathExporter.py; GCodeParameters.py; LinuxCNC.py and it gave me the output that matched my desires more or less. AWJ milling by controlling the jet velocity gives much better results. Of course the efficiency is much lower comparing to conventional milling, but in this case the tool diameter (or milled trench width could be down to 0.2 mm, even less)and the process itself  is clean. I'm trying to find a solution with 0.6.4 version. The principle is quite simple, instead of conventional surfacing, where the tool Z coordinate varies according to the shape, I have to keep the Z constant and to vary Feedrate. Where the speed is high, there the jet erodes very little and where the speed is low the jet erodes deep trench.  for On zigzag corners I have to add the radius (in my case is the NUM controller G-code EB+''radius"), otherwise I have eroded deeper points there. I search the Xmin Xmax if the path grid direction is to the X or Ymin Ymax values if the path grid direction is to the Y on each zigzag row  and add NUM controller G-code fto have a radius. Of course it is enough and it works well for the tests meanwhile the milling contours XY bounds are rectangular...but for not rectangular XY form milling I have to find Xmin Xmax or Ymin Ymax on each zigzag row. The deeper digging in the code is necessary!...and the help is allays welcome! ...what do you think should I open new issue or to reopen the same?