SebKuzminsky/pycam #81: The Engrave/Trace Model field is displayed too small

zancas, 2017-10-22:
<img width="1037" alt="modelunavailable" src="https://user-

The commit I was running against when I generated the above screenshot:

```
commit a815accdc116f032b943b803b1e1e709604686c2 (origin/master, origin/HEAD, master)
Merge: 6269405 e4cb8d4
Author: Sebastian Kuzminsky <seb@highlab.com>
Date:   Mon Oct 16 19:39:44 2017 -0600

    Merge remote-tracking branch 'origin/stable/0.6'
```

zancas, 2017-10-22:
This was originally mentioned in issue:  #79 .   Now that it's tracked here I am editing that issue to remove the redundant data.

closed 2017-11-10 by sumpfralle

sumpfralle, 2017-11-10:
Thank you for the report!

I could not find a good way to adjust the size of the ScrolledWindow element to its children. Thus I defined a fixed width. If anyone knows a better approach, then I would like to improve that.

The issue is fixed in master. It does not apply to stable/0.6.

commit 3e598c28 (2017-11-10): increase the size of model selection lists

By default the ScrolledWindow is too small for being usable.

Closes: #81