SebKuzminsky/pycam #82: At least svg derived representations use sets of lines to represent arcs

zancas, 2017-10-22:


sumpfralle, 2017-11-10:
Yes, currently 2D models use lines as their basic unit. The implemented model operations currently rely on this simplicity.

I can imagine, that arcs will be supported natively by pycam, if we switch somewhen to an external library for 2D operations, that can handle arcs.

Or maybe a configurable precision for pycam's conversion of circles/arcs to lines would be sufficient?

sumpfralle, 2018-09-09:
Recently (f29c90a514828aaeb22d0d8167745fb652b8a591) I added native support for SVG files. Even with this direct import, the resulting items are still lines (even for arcs).
For the forseeable future I do not think, that we will change the internal handling of 2D elements in order to support non-line objects.
Thus I am closing this issue for now.

closed 2018-09-09 by sumpfralle