SebKuzminsky/pycam #84: Relative import syntax enhances the readability of Python code

zancas, 2017-10-22:
This is true because it disambiguates within-package imports from system packages.

We should use relative imports everywhere it makes sense to do so.

sumpfralle, 2017-11-10:
I think, we are approaching the bike-shedding zone :)

In case of doubt, I like to follow [PEP8](https://www.python.org/dev/peps/pep-0008/#imports). Regarding the preferred style of imports it says:

> Absolute imports are recommended, as they are usually more readable and tend to be better behaved (or at least give better error messages) ...
> However, explicit relative imports are an acceptable alternative to absolute imports, especially when dealing with complex package layouts where using absolute imports would be unnecessarily verbose.

Thus I think, absolute imports are quite suitable for pycam.

But maybe someone else has a strong opinion towards a change?

sumpfralle, 2018-08-20:
Closing for now based on the lack of opinions raised to support this change.

closed 2018-08-20 by sumpfralle