SebKuzminsky/pycam #89: Simplify version numbering

pjz, 2017-11-11:
I was going to build a wheel, but...

```
$python2 setup.py
/usr/local/lib/python2.7/dist-packages/setuptools/dist.py:294: User
arning: The version specified (u'0.7.0~pre0.98.g3e598c2') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  "details." % self.metadata.version
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ..
]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: no commands supplied
```

SebKuzminsky, 2017-11-11:
Hi @pjz, the current version numbers are designed to work well as debian package version numbers.  I didn't mean to break them for setup.py in the process, but clearly I did...

My goals for the version number are:
* make version numbers that make sense both on "official" branches (master, stable/*) and on short-lived feature branches
* make version numbers be as nicely human-readable as possible
* have the version number unambiguously identify the git commit that it corresponds to

I wonder if we can come up with something that makes both dpkg and setup.py happy.

[PEP 440](https://www.python.org/dev/peps/pep-0440/)
[Debian package version numbers](https://www.debian.org/doc/debian-policy/#s-f-version)

SebKuzminsky, 2017-11-11:
That seems like it would work.

I would prefer it if the string in all three places "looked" the same, ie using either "dev" or "pre" everywhere.

I have a slight preference for "pre" over "dev",  because to me "pre" seems unambiguously "before", whereas "dev" could be before or after.

So I'm proposing this:
* pre-release tag: "v0.7.0.pre0"
* python setup version "v0.7.0.pre0.98.g3e598c2"
* dpkg version: "v0.7.0~pre0.98.g3e598c2"

sumpfralle, 2017-11-11:
Yes, I would also prefer identical or at least similar version strings.

Thus, I took another look at PEP440:
* indeed `pre` is an allowed tag - indicating a release candidate (synonym for `r` or `rc`)
* non-digits (e.g. for the commit hash) are only allowed for `Local version identifiers`

Thus I would propose the following for a PEP440-compliant version string:
* pre-release tag: `v0.7.0-pre0`
* project version: `v0.7.0-pre0+98.g3e598c2`
* deb package version: `v0.7.0~pre0+98.g3e598c2`

SebKuzminsky, 2017-11-12:
Great!  I'm going to be out of town for a week, but I will work on this when I get back, if no one beats me to it 😊