SebKuzminsky/pycam #92: Segmentation fault with libfreeglut 3.0.0

midyukov-anton, 2017-12-01:
Hi!
When running with libfreeglut-3.0.0 installed, I get a segmentation fault (in the attachment). When using libGLUT-8.0.1 - it works fine. Problem on versions 6.x
[strace_pycam.txt](https://github.com/SebKuzminsky/pycam/files/1521734/strace_pycam.txt)

sumpfralle, 2017-12-02:
Thank you for sharing this information!

Recently pycam switched from the old GTK2/OpenGL library (gtkglext) to GTK3 and GLArea. Thus the problem probably went away. But I would suggest to leave this issue open, until someone confirms that the current master branch works with freeglut-3.0.0 (I cannot test this, since I have only 2.8.1 installed).

midyukov-anton, 2017-12-03:
I checked how it works from the master branch. And with libfreeglut-3.0.0 and c libGLUT I get:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/pycam/Plugins/OpenGLWindow.py", line 677, in paint
    self.glsetup()
  File "/usr/lib/python2.7/site-packages/pycam/Plugins/OpenGLWindow.py", line 454, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)
OpenGL.error.GLError: GLError(
	err = 1282,
	description = 'invalid operation',
	baseOperation = glShadeModel,
	cArguments = (GL_SMOOTH,)
)
```

sumpfralle, 2017-12-03:
Thank you for taking a look at master!

I do not know too much about OpenGL (just enough for fixing trivial things). Do you think that your hardware is well supported?

Could you try to check if the following approach in `pycam/Plugins/OpenGLWindow.py` around line 454 works for you? (you need to take care for proper indentation)
```python
try:
    GL.glShadeModel(GL.GL_SMOOTH)
except:
    pass
```
Do you see visual problems after this change?

midyukov-anton, 2017-12-04:
> Do you see visual problems after this change?

Yes, visual artifacts instead of the image.

> Do you think that your hardware is well supported?

Radeon HD7340, Mesa 17.2.6. Support OpenGL 3.3.

> Could you try to check if the following approach in pycam/Plugins/OpenGLWindow.py around line 454 works for you? (you need to take care for proper indentation)
> 
> try:
>     GL.glShadeModel(GL.GL_SMOOTH)
> except:
>     pass

Did. Another error now:
```
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/pycam/Plugins/OpenGLWindow.py", line 680, in paint
    self.glsetup()
  File "/usr/lib/python2.7/site-packages/pycam/Plugins/OpenGLWindow.py", line 460, in glsetup
    GL.glHint(GL.GL_PERSPECTIVE_CORRECTION_HINT, GL.GL_NICEST)
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)
OpenGL.error.GLError: GLError(
	err = 1280,
	description = 'invalid enumerant',
	baseOperation = glHint,
	cArguments = (
		GL_PERSPECTIVE_CORRECTION_HINT,
		GL_NICEST,
	)
)
```

Same visual artifacts instead of the image.

sumpfralle, 2017-12-04:
> Radeon HD7340, Mesa 17.2.6. Support OpenGL 3.3.

I cannot judge this, but let's just assume, it is well supported.

> Did. Another error now:

If you are in the mood for digging around: please add more try/except clauses around the problematic line above (and all lines that will throw errors later on).

If you manage to get it running and the visualization is somehow usable, then we could integrate these conditional operations into the code.

midyukov-anton, 2017-12-05:
```
If you are in the mood for digging around: please add more try/except clauses around the problematic line above (and all lines that will throw errors later on).

If you manage to get it running and the visualization is somehow usable, then we could integrate these conditional operations into the code.
``` 

Did. Now just a black screen without an image. Probably, it was predictable. I will try on other computers.

sumpfralle, 2018-09-08:
Just a thought: could you try to follow this recommendation: https://github.com/SebKuzminsky/pycam/issues/97#issuecomment-354177779?