SebKuzminsky/pycam #94: Invalid syntax

valeriob01, 2017-12-23:
:~/Downloads/newpycam/pycam/pycam$ ./run_gui.py 
Traceback (most recent call last):
  File "./run_gui.py", line 52, in <module>
    from pycam.Flow.history import DataHistory, merge_history_and_block_events
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/history.py", line 6, in <module>
    from pycam.Flow.parser import dump_yaml, parse_yaml
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/parser.py", line 6, in <module>
    from pycam.Flow.data_models import CollectionName
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/data_models.py", line 438
    yield from cls._get_stable_hashs_for_value(key_value)
             ^
SyntaxError: invalid syntax

sumpfralle, 2017-12-23:
PyCAM started to require python3 instead of python2. The scripts just lacked the proper shebangs. I fixed these now.

Thank you for reporting this!

closed 2017-12-23 by sumpfralle

ebo, 2017-12-23:
Does it require python3, or can it be set up so that it uses both 2 and 
3 with try/except fallback?

On Dec 23 2017 8:29 AM, sumpfralle wrote:
> PyCAM started to require python3 instead of python2. The scripts just
> lacked the proper shebangs. I fixed these now.
>
> Thank you for reporting this!

sumpfralle, 2017-12-23:
There were python2/3 compatible statements up until recently.

Then I asked on the mailing list if someone can imagine a need for python2 compatibility. Noboday raised concerns - thus I was happy to throw away the compatibility code and started to use python3-only syntax features. Now master only runs on python3.

I hope that you don't find a good reason why this could be bad :)

ebo, 2017-12-24:
If I find a good reason that it is bad, then I will be the one to add 
it back in.  I was mostly curious.

Sorry.  Thinking about it a bit, my reaction was more born out of old 
systems that supply python2 only, and various compatibility issues with 
trying to upgrade (think ArcGIS on Win* which hacked versions of the 
libraries to make them work with their proprietary libraries and not 
play nicely with anyone else).  With Anaconda, these compatibility 
issues are mostly a thing of the past.  That said, I still get burned 
from time to time.  I am probably one of the few people on the list that 
pays with metal nights and weekends, and spends the workdays doing 
geospatial programming.  I still write a lot of my stuff in 
compatibility mode...

On Dec 23 2017 1:23 PM, sumpfralle wrote:
> There were python2/3 compatible statements up until recently.
>
> Then I asked on the mailing list if someone can imagine a need for
> python2 compatibility. Noboday raised concerns - thus I was happy to [...]

sumpfralle, 2017-12-24:
I really hope the approaching end-of-life for python2 will put and end to these dependency nightmares quite soon.
Thank you for sharing your thoughts!