SebKuzminsky/pycam #95: error gtk

valeriob01, 2017-12-23:
:~/Downloads/newpycam/pycam/pycam$ ./run_gui.py 
Enabled 4 parallel local processes
Storing a state revision (1/20)

(run_gui.py:12370): Gtk-CRITICAL **: gtk_box_pack: assertion 'gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:12370): Gtk-CRITICAL **: gtk_box_pack: assertion 'gtk_widget_get_parent (child) == NULL' failed
Traceback (most recent call last):
  File "./run_gui.py", line 291, in <module>
    main_func()
  File "./run_gui.py", line 279, in main_func
    exit_code = execute(parser, args, pycam)
  File "./run_gui.py", line 207, in execute
    show_gui()
  File "./run_gui.py", line 131, in show_gui
    plugin_manager.import_plugins()
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/__init__.py", line 299, in import_plugins
    self._load_plugin(plugin, filename, name)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/__init__.py", line 321, in _load_plugin
    if not new_plugin.setup():
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/OpenGLWindow.py", line 160, in setup
    self.area = self._gtk.GLArea(auto_render=False, has_alpha=True, has_depth_buffer=True)
  File "/usr/lib/python3/dist-packages/gi/module.py", line 320, in __getattr__
    return getattr(self._introspection_module, name)
  File "/usr/lib/python3/dist-packages/gi/module.py", line 139, in __getattr__
    self.__name__, name))
AttributeError: 'gi.repository.Gtk' object has no attribute 'GLArea'

sumpfralle, 2017-12-23:
Which version of GTK is installed on your system?

valeriob01, 2017-12-23:
> Which version of GTK is installed on your system?

the latest on debian jessie

sumpfralle, 2017-12-24:
I am sorry, but Debian jessie contains only GTK 3.14 - but the GLArea widget is supported only since v3.16. I clarified this now in the requirements documentation and added a warning (instead of the stacktrace you encountered).

I am afraid, that you will need to stick to PyCAM v0.6.x until you can upgrade your system to the current stable Debian release.

valeriob01, 2017-12-24:
I have only installed gedit which installs python3-cairo and python3-gi-cairo as dependencies:

[image]

sumpfralle, 2017-12-24:
The problem is probably your version of `gir1.2-gtk-3.0` (i.e. GTK). In Debian Jessie it is v3.14. Debian Stretch contains v3.22.

The current code in master requires at least v3.16. Thus Debian Jessie (oldstable) is not sufficient.

sumpfralle, 2017-12-26:
I assume that the stacktrace you posted does not appear with GTK v3.16 or later.

closed 2017-12-26 by sumpfralle