SebKuzminsky/pycam #97: glShadeModel: invalid operation -> no 3d

FDePourcq, 2017-12-23:
Hi, 
when trying out the master branch ( 23 dec 2017 ) I get no opengl rendering and lots of errors are thrown:
```
$ ./pycam/run_gui.py 
Enabled 8 parallel local processes
Storing a state revision (1/20)

(run_gui.py:17164): Gtk-WARNING **: Theme parsing error: gtk.css:68:35: The style property GtkButton:child-displacement-x is deprecated and shouldn't be used anymore. It will be removed in a future version

(run_gui.py:17164): Gtk-WARNING **: Theme parsing error: gtk.css:69:35: The style property GtkButton:child-displacement-y is deprecated and shouldn't be used anymore. It will be removed in a future version

(run_gui.py:17164): Gtk-WARNING **: Theme parsing error: gtk.css:73:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version

(run_gui.py:17164): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:17164): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 718, in paint
    self.glsetup()
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 498, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 402, in __call__
    return self( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Font directory: /home/donf/src_other/pycam/share/fonts
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 498, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 498, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 498, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 498, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Loading workspace from file: /home/donf/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
Imported STL model: 0 vertices, 0 edges, 12 triangles
Imported 1 items into 'export_settings'
Storing a state revision (2/20)
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 718, in paint
    self.glsetup()
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 498, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
```

I am running debian:
```
$ uname -a
Linux XIV 4.13.0-1-amd64 #1 SMP Debian 4.13.13-1 (2017-11-16) x86_64 GNU/Linux
$ dpkg -l | grep -i python | grep -i -e opengl -e gtk
ii  python-glade2                                 2.24.0-5.1+b1                          amd64        GTK+ bindings: Glade support
ii  python-gtk2                                   2.24.0-5.1+b1                          amd64        Python bindings for the GTK+ widget set
ii  python-pyside.qtopengl                        1.2.2-2+b1                             amd64        Qt 4 OpenGL module - Python bindings
ii  python-qt4-gl                                 4.12.1+dfsg-2                          amd64        Python bindings for Qt4's OpenGL module
ii  python-spice-client-gtk                       0.30-1                                 amd64        GTK2 widget for SPICE clients (Python binding)
ii  python-webkit                                 1.1.8-3.1                              amd64        WebKit/Gtk Python bindings
ii  python-wxgtk3.0                               3.0.2.0+dfsg-6                         amd64        Python interface to the wxWidgets Cross-platform C++ GUI toolkit
ii  python3-opengl                                3.1.0+dfsg-1                           all          Python bindings to OpenGL (Python 3)
```

In pycam version 6.2 the opengl-part works fine on my system. I might dig in to it because I want good toolpaths. :)

FDePourcq, 2017-12-23:
https://github.com/SebKuzminsky/pycam/issues/92  :  sumpfralle is hitting the same issue.

FDePourcq, 2017-12-24:
It seems to be some kind of linker-issue ... as if the python wrapper for opengl knows about symbols but can not find a single one. And each time such a symbol is used it throws errors like "invalid operation" "invalid enumerant" instead.

sumpfralle, 2017-12-24:
Did you try to run the previous state (e.g. commit 77dd887e17d1b2c7deb79cd524d5277f86857df1) with python3 (e.g. `python3 pycam/run_gui.py`)? It uses the same approach with regard to OpenGL - thus I would expect, that it fails, too.

From a distance, this feels like a local library issue. Even though I would not expect this with a Debian system (I hardly ever encountered this). Maybe you are using external repositories (e.g. deb-multimedia)?

Regarding #92: the submitter of that bug report noticed a difference between libglut (works) and libfreeglut (fails). Did you try this, as well?

FDePourcq, 2017-12-24:
77dd887 with `python3 pycam/run_gui.py` has it too.
It has been a while since I worked in python but I noticed that all GL-related imports where done in __init__.py and the reference to those imports is being passed around (as if a new import would instantiate something new, instead of simply redoing the import), I wonder if that approach to imports has something to do with it.

FDePourcq, 2017-12-24:
After fiddling some more with the code , by redoing the imports stumbled on something weird:
```
Traceback (most recent call last):
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 740, in paint
    self.glsetup()
  File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", line 514, in glsetup
    | GLUT.GLUT_MULTISAMPLE | GLUT.GLUT_ALPHA | GLUT.GLUT_ACCUM)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 361, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 366, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/lib/x86_64-linux-gnu/libglut.so.3: undefined symbol: GLUT_RGBA
Font directory: /home/donf/src_other/pycam/share/fonts
```
The python code referenced `GLUT.GLUT_RGBA` and this value is defined with a preprocessor #define in glut, so it should not result in compiled code, it should not be present as a symbol in the dynamic library (.so). 
And indeed I see in PyOpenGl in raw/GLUT/constants.py  the thing explicitly redefined for python: 
`GLUT_RGBA = Constant( 'GLUT_RGBA', 0)` ... but still for unknown reason ... that constant is being searched in the dynamic library ... 
I suspect that the root-cause of that problem will be really close to the root-cause of the other problems here.

ebo, 2017-12-24:
Regardless of how you/we sort this out, I think it would be a good idea 
to post a report upstream.

On Dec 24 2017 9:54 AM, FDePourcq wrote:
> After fiddling some more with the code , by redoing the imports
> stumbled on something weird:
> ```
> Traceback (most recent call last):
>   File "/home/donf/src_other/pycam/pycam/Plugins/OpenGLWindow.py", [...]

sumpfralle, 2017-12-24:
> It has been a while since I worked in python but I noticed that all GL-related imports where done in init.py and the reference to those imports is being passed around (as if a new import would [...]

I used this approach in order to avoid rather complicated (and repetitive) conditional imports (since OpenGL is not strictly required). I doubt, that this import-once approach could change the behaviour of the OpenGL libraries - but of course, there could be magic involved :)

I still find it a bit puzzling, that you could encounter such issues with a clean Debian installation without external repositories. I am also using Debian "testing" (just for i386 instead of amd64).

If your local installation is fine, then I agree with ebo, that an upstream bug report could be the way to proceed.

FDePourcq, 2017-12-25:
It is not a clean installation in the sense that it has gone through several upgrades already. So a completely clean installation might not have these problems. I will give it some more time and if that does not work ... will look into that mysterious upstream. (Whatever that means: i don't see a debian package for pycam, i can't say if it is an issue with python3 or with pyopengl or with debian packaging or with some extra layer i don't know about yet.)

sumpfralle, 2017-12-25:
> It is not a clean installation in the sense that it has gone through several upgrades already.

I did not want to appear so picky :)

An upgraded installation is perfectly fine.
I just wanted to point out that third-party repositories (most prominently: deb-multimedia.org) can cause unwanted side-effects (especially if the epoch of the package version was increased - thus stopping regular upgrades from happening even after the repository was removed). A quick `dpkg -l | grep dmo` should probably expose most candidates for such a situation.

Maybe you can use `debootstrap` for a quick test with a (quite) clean environment? If this fails too, I would suspect a hardware/driver issue.

valeriob01, 2017-12-25:
Debian 9.3:

$ ./run_gui.py 
Enabled 4 parallel local processes
Storing a state revision (1/20)

(run_gui.py:1172): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:1172): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed
Traceback (most recent call last):
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/OpenGLWindow.py", line 725, in paint
    self.glsetup()
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/OpenGLWindow.py", line 505, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 402, in __call__
    return self( *args, **named )
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glShadeModel,
	cArguments = (GL_SMOOTH,)
)
Font directory: /home/sel/Downloads/newpycam/pycam/share/fonts
No preferences file found ([Errno 2] No such file or directory: '/home/sel/.pycam/preferences.conf'). Starting with default preferences.
Loading workspace from file: /home/sel/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
STLImporter: Failed to read file (/home/sel/Downloads/newpycam/pycam/pycam/samples/Box0.stl): [Errno 2] No such file or directory: '/home/sel/Downloads/newpycam/pycam/pycam/samples/Box0.stl'
Traceback (most recent call last):
  File "./run_gui.py", line 295, in <module>
    main_func()
  File "./run_gui.py", line 283, in main_func
    exit_code = execute(parser, args, pycam)
  File "./run_gui.py", line 211, in execute
    show_gui()
  File "./run_gui.py", line 135, in show_gui
    gui.load_startup_workspace()
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/__init__.py", line 215, in load_startup_workspace
    remember_uri=False, default_content=DEFAULT_WORKSPACE)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/__init__.py", line 243, in load_workspace_from_file
    self.load_workspace_from_description(content)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/__init__.py", line 278, in load_workspace_from_description
    reset=True)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/parser.py", line 54, in parse_yaml
    if item_class(name, data) is None:
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/data_models.py", line 739, in __init__
    self.get_collection().append(self)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/data_models.py", line 649, in append
    self.notify_list_changed()
  File "/home/sel/Downloads/newpycam/pycam/pycam/Flow/data_models.py", line 717, in notify_list_changed
    get_event_handler().emit_event(self._list_changed_event)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Utils/events.py", line 101, in emit_event
    handler.func(*(handler.args + args), **kwargs)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/__init__.py", line 171, in <lambda>
    result = lambda *args: self.core.emit_event(func, *params)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Utils/events.py", line 101, in emit_event
    handler.func(*(handler.args + args), **kwargs)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/OpenGLViewDimension.py", line 52, in update_model_dimensions
    model_box = pycam.Geometry.Model.get_combined_bounds(models)
  File "/home/sel/Downloads/newpycam/pycam/pycam/Geometry/Model.py", line 43, in get_combined_bounds
    low[0] = model.minx
AttributeError: 'NoneType' object has no attribute 'minx'

FDePourcq, 2017-12-25:
- The additional issue from valeriob01 (`AttributeError: 'NoneType' object has no attribute 'minx'`) might have something to do with a model that is being loaded at startup because i don't see that one.

- My sources.list never needed special repositories, because of this issue however i'm fiddling with that pip3-package manager.

Also, i rebooted the same machine to a ubuntu (which also never saw sources.list-modifications and even less upgrades): 
- that one hits a pyopengl bug `AttributeError: module 'OpenGL.GL' has no attribute 'GL_READ_WRITE'`
- after removing the python3-opengl debian package and doing `pip3 install --upgrade PyOpenGL` (as user) 
  i got in the same  mess as in Debian: 
```
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
```

valeriob01, 2017-12-25:
from https://www.khronos.org/registry/OpenGL-Refpages/gl2.1/xhtml/glShadeModel.xml

GL_INVALID_OPERATION is generated if glShadeModel is executed between the execution of glBegin and the corresponding execution of glEnd.

unrealsolver, 2017-12-25:
I have the same issue using this commit:
```
commit 69a78b319006ee74ec6aaf1de6dd0862cb9da838 (HEAD)
Author: Lars Kruse <devel@sumpfralle.de>
Date:   Sat Nov 25 18:05:49 2017 +0100
```
Probably, the problem was introduced in 3p dependency. It's sad, there are no unit tests.

sumpfralle, 2017-12-25:
@valeriob01: the error handling while loading the workspace file is improved now. Thus the one error (due to OpenGL) will not be mangled with unrelated error output anymore.

@FDePourcq: thank you for taking the time to try it in a clear environment. What a pity, that this did not offer any new clues to us (as far as I understood).

@unrealsolver: yes, frozen requirements would prevent a certain class of problems with the cost of introducing another one. Personally I prefer non-versioned dependencies.

@ all: could you please try if the same error happens with commit 77dd887e17d1b2c7deb79cd524d5277f86857df1 - either with python2 or python3?
```
python pycam/run_gui.py
python3 pycam/run_gui.py
```

I am wondering if this is a python3-opengl-specific issue (but I have no idea at all).
Any other ideas how to debug this further?

valeriob01, 2017-12-25:
Now there is a blank 3d model window

[image]

but still:
   baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glShadeModel,
	cArguments = (GL_SMOOTH,)
)

sumpfralle, 2017-12-25:
> Now there is a blank 3d model window

under which circumstances?

valeriob01, 2017-12-25:
python3 pycam/run_gui.py

unrealsolver, 2017-12-25:
The error is still the same on my non-debian os.
I wrote small program, using this OpenGL python library, it works without crashes, when calling glShadeModel, but I have no experience with GL and have no idea how things work.

sumpfralle, 2017-12-25:
@valeriob01 & @unrealsolver : did you use the current master for your test? Did you also try the commit 77dd887e17d1b2c7deb79cd524d5277f86857df1 with python2 and python3? What exactly was the behaviour of these three situations?
* HEAD of master -> python3
* 77dd887e17d1b2c7deb79cd524d5277f86857df1 -> python2
* 77dd887e17d1b2c7deb79cd524d5277f86857df1 -> python3

Thank you!

unrealsolver, 2017-12-25:
The output for python3 and master / https://github.com/SebKuzminsky/pycam/commit/77dd887e17d1b2c7deb79cd524d5277f86857df1 are same (the OpenGL error part is exact same). I haven't installed python2 deps yet and couln't test it yet.

FDePourcq, 2017-12-25:
not getting very far.

I figured that if you put `GLUT.glutCreateWindow('Useless window for testing only.')` before the first glShadeModel  that most errors are not thrown anymore. So that call, besides making a window, must be doing some magic behind the scenes to direct gl-calls to its new window ... probably something with a opengl-context (?)
So I suspect that ... those gl-calls do not have a required reference to the opengl-context of the gtkglarea  ? 
Stuff like 'self.area.get_context().make_current()` does not seem to help :(

But the demo-scripts in pyopengl seem to work (after converting them from python2 to python3 using the script 2to3-3.6 )...  of course those demo-scripts have no gtkglarea...

FDePourcq, 2017-12-25:
I think i know what is going on: 
python-gtkglext1 might not have been updated to support python3 ...
The last commit in the underlying project pygtkglext was in 2010 :)
The last commit in the debian changelog for that package was in 2012 :)
Pycam 6.x used it.
python-gtkglext1 contains stuff for python 2.6 and python2.7 but that's it.

--> ... even if i figure out how to build it for python3 ... ... is it possible to resurrect the living dead debian package maintainers?

sumpfralle, 2017-12-26:
@FDePourcq: almost :)

PyCAM switched from GTK2 to GTK3 and from Python2 to Python3. This also included the switch from gtkglext (gtk2/python2) to gtk.GLArea (GTK3). Thus gtkglext is finally not an issue anymore.

I am very confused, that the current state is not working for the three of you.
Thus let us compare our installed libraries:

```
$ dpkg -l |grep -E "(glut|python3?-opengl|gir1.2-gtk-3.0|libglu?1)"
ii  freeglut3:i386                                              2.8.1-3                             i386         OpenGL Utility Toolkit
ii  gir1.2-gtk-3.0:i386                                         3.22.24-3                           i386         GTK+ graphical user interface library -- gir bindings
ii  libgl1:i386                                                 1.0.0-1                             i386         Vendor neutral GL dispatch library -- legacy GL support
ii  libgl1-mesa-dri:i386                                        17.2.5-1                            i386         free implementation of the OpenGL API -- DRI modules
ii  libgl1-mesa-glx:i386                                        17.2.5-1                            i386         transitional dummy package
ii  libglu1-mesa:i386                                           9.0.0-2.1                           i386         Mesa OpenGL utility library (GLU)
ii  python-opengl                                               3.1.0+dfsg-1                        all          Python bindings to OpenGL (Python 2)
ii  python3-opengl                                              3.1.0+dfsg-1                        all          Python bindings to OpenGL (Python 3)
```

With the libraries above, the current master works with Python3 and 77dd887e17d1b2c7deb79cd524d5277f86857df1 work with Python2 and Python3 on my system. Please make sure that you do not use manually compiled libraries (e.g. via virtualenv), when checking the same on your systems.

FDePourcq, 2017-12-26:
```
$ dpkg -l |grep -E "(glut|python3?-opengl|gir1.2-gtk-3.0|libglu?1)"
ii  freeglut3:amd64                               2.8.1-3                                amd64        OpenGL Utility Toolkit
ii  freeglut3-dev:amd64                           2.8.1-3                                amd64        OpenGL Utility Toolkit development files
ii  gir1.2-gtk-3.0:amd64                          3.22.24-3                              amd64        GTK+ graphical user interface library -- gir bindings
ii  libgl1:amd64                                  1.0.0-1                                amd64        Vendor neutral GL dispatch library -- legacy GL support
ii  libgl1:i386                                   1.0.0-1                                i386         Vendor neutral GL dispatch library -- legacy GL support
ii  libgl1-mesa-dev:amd64                         17.2.5-1                               amd64        free implementation of the OpenGL API -- GLX development files
ii  libgl1-mesa-dri:amd64                         17.2.5-1                               amd64        free implementation of the OpenGL API -- DRI modules
ii  libgl1-mesa-dri:i386                          17.2.5-1                               i386         free implementation of the OpenGL API -- DRI modules
ii  libgl1-mesa-glx:amd64                         17.2.5-1                               amd64        transitional dummy package
ii  libgl1-mesa-glx:i386                          17.2.5-1                               i386         transitional dummy package
ii  libgl1-nvidia-glx:amd64                       384.98-3                               amd64        NVIDIA binary OpenGL/GLX library (non-GLVND variant)
ii  libglu1-mesa:amd64                            9.0.0-2.1                              amd64        Mesa OpenGL utility library (GLU)
ii  libglu1-mesa:i386                             9.0.0-2.1                              i386         Mesa OpenGL utility library (GLU)
ii  libglu1-mesa-dev:amd64                        9.0.0-2.1                              amd64        Mesa OpenGL utility library -- development files
ii  libkwinglutils11                              4:5.10.5-2+b1                          amd64        KDE window manager gl utils library
ii  libkwinglutils1abi2                           4:4.11.22-3                            amd64        library with OpenGL utilities for the KDE window manager
ii  python-opengl                                 3.1.0+dfsg-1                           all          Python bindings to OpenGL (Python 2)
ii  python3-opengl                                3.1.0+dfsg-1                           all          Python bindings to OpenGL (Python 3)
```

unrealsolver, 2017-12-26:
```
pacman -Q |grep -E "(glut|python3?-opengl|gir1.2-gtk-3.0|libglu?1|gobject)"
freeglut 3.0.0-2
gobject-introspection-runtime 1.54.1-1
pygobject-devel 3.26.1-1
pygobject2-devel 2.28.6-13
python-gobject 3.26.1-1
python-opengl 3.1.0-3
python2-gobject2 2.28.6-13
```

valeriob01, 2017-12-26:
$ dpkg -l |grep -E "(glut|python3?-opengl|gir1.2-gtk-3.0|libglu?1)"
ii  freeglut3:amd64                       2.8.1-3                                     amd64        OpenGL Utility Toolkit
ii  gir1.2-gtk-3.0:amd64                  3.22.11-1                                   amd64        GTK+ graphical user interface library -- gir bindings
ii  libgl1-mesa-dri:amd64                 13.0.6-1+b2                                 amd64        free implementation of the OpenGL API -- DRI modules
ii  libgl1-mesa-glx:amd64                 13.0.6-1+b2                                 amd64        free implementation of the OpenGL API -- GLX runtime
ii  libglu1-mesa:amd64                    9.0.0-2.1                                   amd64        Mesa OpenGL utility library (GLU)
ii  python3-opengl                        3.1.0+dfsg-1                                all          Python bindings to OpenGL (Python 3)

FDePourcq, 2017-12-26:
```
			F			V		US		SF
freeglut3:amd64		2.8.1-3			2.8.1-3		3.0.0-2		
gir1.2-gtk-3.0:amd64	3.22.24-3		3.22.11-1	/		
libgl1-mesa-dri:amd64	17.2.5-1		13.0.6-1+b2	/		
libgl1-mesa-dri:i386                                        			17.2.5-1
libgl1-mesa-glx:amd64	17.2.5-1		13.0.6-1+b2	/		
libgl1-mesa-glx:i386                                        			17.2.5-1   
libglu1-mesa:amd64	9.0.0-2.1		9.0.0-2.1	/		
libglu1-mesa:i386                               /            	/		9.0.0-2.1
python3-opengl		3.1.0+dfsg-1		3.1.0+dfsg-1	3.1.0-3		3.1.0+dfsg-1 
python-opengl		3.1.0+dfsg-1		/		/		3.1.0+dfsg-1
```

valeriob01, 2017-12-26:
Just a guess, probably GL.glShadeModel is a deprecated function ?

FDePourcq, 2017-12-26:
no, because the other gl-calls have it too and they do work after glut.createwindow. My guess is that the implementation of gtk.glarea does not succeed in making itself the "current opengl context" -> the subsequent opengl-calls are no valid operations on a non-existing context.

valeriob01, 2017-12-26:
Which other gl-calls ?

valeriob01, 2017-12-26:
I have commented out the glShadeModel and this is the resulting error:

   GL.glHint(GL.GL_PERSPECTIVE_CORRECTION_HINT, GL.GL_NICEST)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 232, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glHint,
	cArguments = (
		GL_PERSPECTIVE_CORRECTION_HINT,
		GL_NICEST,
	)
)

unrealsolver, 2017-12-26:
@valeriob01 That's right, I also tried this trick. As @FDePourcq mentioned, it's most probably gtk/GL(UT) initialization error. We should inspect calls chain of related code to figure out where the actual exception occur. `invalid operation/enumerant` and other errors are just consequences.

FDePourcq, 2017-12-26:
Crawling through the rabbit hole here:
`apt-get source libgtk-3-0` --> this gets all code for gtk (several packages in `gtk+3.0-3.22.24` )
`debuild -i -us -uc -b` --> this builds the packages
but `cp ./debian/build/deb/gtk/.libs/libgtk-3.so.0.2200.24 /usr/lib/x86_64-linux-gnu/libgtk-3.so.0.2200.24`  (for gtk)
and `cp ./debian/build/deb/gdk/.libs/libgdk-3.so.0.2200.24 /usr/lib/x86_64-linux-gnu/libgdk-3.so.0.2200.24` (for gdk) deploys faster, no need to wait for all the packages and installing them.

For now i see nothing out of the expected: `gtk_gl_area_make_current` is called shortly before pycam starts doing gl-calls, `gtk_gl_area_make_current` properly calls `gdk_gl_context_make_current` , ...

`gdk_gl_context_realize` was also fired properly :(  ... gonna have to ... dig deeper ....

valeriob01, 2017-12-27:
@FDePourcq Not sure but gtk.GLArea is not supposed to make itself the current context, you have to call https://developer.gnome.org/gtk3/stable/GtkGLArea.html#gtk-gl-area-make-current afterwards.

FDePourcq, 2017-12-27:
i tried things like `self.area.get_context().make_current()` , didn't help, this needs to go upstream indeed.

sumpfralle, 2017-12-27:
I am amazed, that you were (probably painfully) digging deeper!

I am still confused, that everything is working fine for me - while you are having issues. This should indicate that it is indeed not a high-level problem (e.g. caused by the way the PyCAM code handles GLArea). Or what do you think?

I will ask on the [pycam-devel mailinglist](https://sourceforge.net/p/pycam/mailman/message/36170080/) if it works for them. Maybe this will give us a better feeling for the source of the problem.

FDePourcq, 2017-12-27:
So I gave it one more try: 
in libepoxy i see that `epoxy_internal_has_gl_extension` is called, with a demand for `GL_EXT_framebuffer_object`. This one it can not find. It is requested 5 times, to no avail.
Opengl version (in the epoxy logic) is `45`. Or 4.50 in human speak.

"maybe my system does not have that specific extension??"
```
$ glxinfo | grep -e "version" -e GL_EXT_framebuffer_object -e GL_EXT_framebuffer_sRGB
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL core profile version string: 4.5 (Core Profile) Mesa 17.2.5
OpenGL core profile shading language version string: 4.50
    GL_EXT_framebuffer_sRGB, GL_EXT_packed_depth_stencil, GL_EXT_packed_float, 
OpenGL version string: 3.0 Mesa 17.2.5
OpenGL shading language version string: 1.30
    GL_EXT_framebuffer_object, GL_EXT_framebuffer_sRGB, 
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 17.2.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```
It has it but ... not in the list directly beneath the opengl 4.50 ... it is only mentioned in the list beneath 
```
OpenGL version string: 3.0 Mesa 17.2.5
OpenGL shading language version string: 1.30
```

So ... either ... my opengl 4.50 should have that `GL_EXT_framebuffer_object`  or ... i need another opengl version?

Anyway ... getting closer.

FDePourcq, 2017-12-27:
WAAAA IT WOOOORKSS
```
MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"  python3 ./pycam/run_gui.py
```

ebo, 2017-12-27:
Woo Hoo!  Congrats ;-)

On Dec 27 2017 2:07 PM, FDePourcq wrote:
> WAAAA IT WOOOORKSS
> ```
> MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"  python3 
> ./pycam/run_gui.py
> ```

FDePourcq, 2017-12-27:
oh wait it is no solution yet, the different mesa-version offers a useless gui, i am figuring out if i can't get different graphics-drivers.

ebo, 2017-12-27:
I will still give you a "Woo Hoo"... we need to celibate our successes 
from time to time, even when they are small  ;-)

On Dec 27 2017 3:17 PM, FDePourcq wrote:
> oh wait it is no solution yet, the different mesa-version offers a
> useless gui, i am figuring out if i can't get different
> graphics-drivers.

valeriob01, 2017-12-28:
SBANG !!! Good!

$ MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5" python3 pycam/run_gui.py 
Enabled 4 parallel local processes
Storing a state revision (1/20)

(run_gui.py:1328): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:1328): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed
Font directory: /home/sel/Downloads/newpycam/pycam/share/fonts
Loading workspace from file: /home/sel/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
Imported STL model: 0 vertices, 0 edges, 12 triangles
Imported 1 items into 'export_settings'

(run_gui.py:1328): Gtk-WARNING **: Allocating size to GtkBox 0x56312da33170 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?

(run_gui.py:1328): Gtk-WARNING **: Allocating size to GtkWindow 0x56312dc1a2a0 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?

(run_gui.py:1328): Gtk-WARNING **: Allocating size to GtkBox 0x56312da33170 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?

(run_gui.py:1328): Gtk-WARNING **: Allocating size to GtkWindow 0x56312dc1a2a0 without calling gtk_widget_get_preferred_width/height(). How does the code know the size to allocate?
Storing a state revision (2/20)

[image]

unrealsolver, 2017-12-29:
Mesa override trick does not work for me either. In addition, have following error:
```
Traceback (most recent call last):
  File "/home/ruslan/dev/pycam/pycam/Plugins/OpenGLWindow.py", line 743, in paint
    self.core.emit_event("visualize-items")
  File "/home/ruslan/dev/pycam/pycam/Utils/events.py", line 101, in emit_event
    handler.func(*(handler.args + args), **kwargs)
  File "/home/ruslan/dev/pycam/pycam/Plugins/OpenGLViewSupportModelPreview.py", line 45, in draw_support_preview
    models = self.core.get("current_support_models")[:]
TypeError: 'NoneType' object is not subscriptable
```

sumpfralle, 2017-12-30:
> In addition, have following error: [..]

fixed now - see 40dc12f812532ff79b0809bdf7d49bfcf37ad1ff

sumpfralle, 2018-02-15:
@FDePourcq, @valeriob01, @unrealsolver: what is the current status of this bug?

FDePourcq, 2018-02-15:
I just did a git pull and ./pycam/run_gui.py  --> still the same.

I ... moved on ... by writing my own script for getting that drill perfect the way i wanted it ... i should still upload that script to github.  (Also i got myself a 4th axis -> pycam would not have helped me there anyway.)

tracek, 2018-03-16:
Hi - I am having exactly same issue.

```
Storing a state revision (2/20)
Traceback (most recent call last):
  File "/home/tracek/Libraries/pycam/pycam/Plugins/OpenGLWindow.py", line 726, in paint
    self.glsetup()
  File "/home/tracek/Libraries/pycam/pycam/Plugins/OpenGLWindow.py", line 506, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 208, in glCheckError
    baseOperation = baseOperation,
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glShadeModel,
	cArguments = (GL_SMOOTH,)
```

When running with `MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5"  python3 run_gui.py`, I get the following:

```
Enabled 4 parallel local processes
Storing a state revision (1/20)

(run_gui.py:29679): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:29679): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed
Traceback (most recent call last):
  File "/home/tracek/Libraries/pycam/pycam/Plugins/OpenGLWindow.py", line 726, in paint
    self.glsetup()
  File "/home/tracek/Libraries/pycam/pycam/Plugins/OpenGLWindow.py", line 499, in glsetup
    GLUT.glutInit()
  File "/usr/lib/python3/dist-packages/OpenGL/GLUT/special.py", line 318, in glutInit
    holder[i] = arg
TypeError: bytes or integer address expected instead of str instance
Font directory: /home/tracek/Libraries/pycam/share/fonts
Loading workspace from file: /home/tracek/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
Imported STL model: 12 triangles
Imported 1 items into 'export_settings'
Storing a state revision (2/20)
freeglut  ERROR:  Function <glutWireCone> called without first calling 'glutInit'.
```

Any help would be appreciated!

sumpfralle, 2018-03-17:
@tracek: I took a  look at the code that could lead to the error thrown above by `GLUT.glutInit()`.

Here on my system (Debian testing, i386, python3-openg==3.1.0+dfsg-1), the above error cannot be thrown, as far as I can tell (all types are properly handled in that function).

Which version of python3-opengl do you use?

tracek, 2018-03-17:
@sumpfralle Thanks for checking it. I am on 3.0.2-1 (latest in the stable repo), Xubuntu 16.04.

esuah-net, 2018-07-13:
Hi. Same here with 0.7.0~pre0.480.gd0d0091.dirty, Ubuntu Mate 18.04, Nvidia GeForce GTX 750 Ti, Driver 396.24.02:

OpenGL.error.GLError: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glHint,
	cArguments = (
		GL_PERSPECTIVE_CORRECTION_HINT,
		GL_NICEST,
	)
)

To verify my driver setup i tried this example:
http://code.activestate.com/recipes/325391-open-a-glut-window-and-draw-a-sphere-using-pythono/
It works an gives no error when I put
glHint(GL_PERSPECTIVE_CORRECTION_HINT, GL_NICEST)
in line 13 after
glutCreateWindow(name)

I don't know if this is enough to verify that the drivers works.

Because my python skills are not so good yet the code is not easy to debug for me.
I can't find something like a render context/surface/buffer in the code.
Digging on...

sumpfralle, 2018-07-13:
My OpenGL understand is quite minimal.
Thus please confirm, whether you are suggesting the following flow:
1. put `glHint(GL_PERSPECTIVE_CORRECTION_HINT, GL_NICEST)` in a try/except block at the beginning of the 3D view initialization (for checking the usability of the local graphic driver)
2. emit a warning (and disable 3D visualization) in case of an error?

emogenet, 2018-07-20:
A few comments:
   - The issue still exists with HEAD d0d0091014f518e6d193ac153028967a786fd3f9
   - I am experiencing the problem on a clean lucid (17.10) Ubuntu release with an Intel HD Graphics 620 chip. That should be hopefully be easy to reproduce.
   - The good news : I used the export MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5" trick and pycam now seems to works for me

hugovangalen, 2018-10-19:
FYI, I am having this issue on Ubuntu 16.04.5 LTS on a set-up with two graphic adapters: an Intel Xeon E3-1200 (i915 kernel module) and an AMD Radeon HD 7770/8760 (radeon kernel module), kernel 4.4.0-137-generic, python3 (3.5.2), pyopengl (3.0.2-1).

The MESA_GL_VERSION_OVERRIDE trick does not work for me. It shows "Unable to create GL context" or something similar where the visualisation should show.

sumpfralle, 2018-12-16:
I think, the issue is caused by #134.

vladtcvs, 2019-02-19:
Problem still present. Debian testing, mesa 18.3.2

mesa info:
```
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: X.Org (0x1002)
    Device: AMD CAICOS (DRM 2.49.0 / 4.9.0-8-amd64, LLVM 7.0.1) (0x6779)
    Version: 18.3.2
    Accelerated: yes
    Video memory: 1024MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
```

```
OpenGL vendor string: X.Org
OpenGL renderer string: AMD CAICOS (DRM 2.49.0 / 4.9.0-8-amd64, LLVM 7.0.1)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.3.2
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
```

MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5" fixes the problem

vladtcvs, 2019-02-19:
After running with gl override, I started without it, and now 3d was working, but errors 
```
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
``` were present

sumpfralle, 2019-02-19:
Please note that this issue will probably never get fixed (see #134), unless someone with a good amount of OpenGL-specific knowledge wants to migrate the old code.

Everyone else: please try the branch `replace-opengl-with-x3d` (see #135) and report any issues you encounter.

closed 2019-02-19 by sumpfralle