SebKuzminsky/pycam #98: emty canvas: no drawings after model is loaded

cki61, 2017-12-27:
Hi all,

independent from the "save workspace" updates I only see an empty canvas after the model is loaded. Please see the very long log from starting App by command line.
I use openSUSE 42.3 with all necessary python packages.
Maybe someone gets an idea whats wrong with my installation or the app.

Regards,
cki61

`cki@ckilap02:~/D_IWS022/bin/pycam> pycam/run_gui.py 

(run_gui.py:7242): Gtk-WARNING **: Theme parsing error: gtk.css:68:35: The style property GtkButton:child-displacement-x is deprecated and shouldn't be used anymore. It will be removed in a future version

(run_gui.py:7242): Gtk-WARNING **: Theme parsing error: gtk.css:69:35: The style property GtkButton:child-displacement-y is deprecated and shouldn't be used anymore. It will be removed in a future version

(run_gui.py:7242): Gtk-WARNING **: Theme parsing error: gtk.css:73:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version
Enabled 4 parallel local processes
Storing a state revision (1/20)

(run_gui.py:7242): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed

(run_gui.py:7242): Gtk-CRITICAL **: gtk_box_pack: assertion '_gtk_widget_get_parent (child) == NULL' failed                                                                                                                                
Traceback (most recent call last):                                                                                                                                                                                                         
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 725, in paint                                                                                                                                                    
    self.glsetup()                                                                                                                                                                                                                         
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 505, in glsetup                                                                                                                                                  
    GL.glShadeModel(GL.GL_SMOOTH)                                                                                                                                                                                                          
  File "/usr/lib/python3.4/site-packages/OpenGL/platform/baseplatform.py", line 402, in __call__                                                                                                                                           
    return self( *args, **named )                                                                                                                                                                                                          
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)                                                                                                                 
OpenGL.error.GLError: GLError(                                                                                                                                                                                                             
        err = 1282,                                                                                                                                                                                                                        
        description = b'invalid operation',                                                                                                                                                                                                
        baseOperation = glShadeModel,                                                                                                                                                                                                      
        cArguments = (GL_SMOOTH,)                                                                                                                                                                                                          
)                                                                                                                                                                                                                                          
Font directory: /home/cki/D_IWS022/bin/pycam/share/fonts                                                                                                                                                                                   
Loading workspace from file: /home/cki/.pycam/workspace.yml
Imported 2 items into 'tools'
Imported 2 items into 'processes'
Imported 1 items into 'bounds'
Imported 2 items into 'tasks'
Imported 1 items into 'models'
Imported STL model: 0 vertices, 0 edges, 12 triangles
Imported 1 items into 'export_settings'
Traceback (most recent call last):
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 725, in paint
    self.glsetup()
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 505, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Traceback (most recent call last):
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 725, in paint
    self.glsetup()
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 505, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Traceback (most recent call last):
[... 93 lines cut ...]
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 725, in paint
    self.glsetup()
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 505, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
Traceback (most recent call last):
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 725, in paint
    self.glsetup()
  File "/home/cki/D_IWS022/bin/pycam/pycam/Plugins/OpenGLWindow.py", line 505, in glsetup
    GL.glShadeModel(GL.GL_SMOOTH)
  File "errorchecker.pyx", line 53, in OpenGL_accelerate.errorchecker._ErrorChecker.glCheckError (src/errorchecker.c:1218)
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glShadeModel,
        cArguments = (GL_SMOOTH,)
)
cki@ckilap02:~/D_IWS022/bin/pycam> 
`

sumpfralle, 2017-12-28:
This looks similar to issue #97.

Please try the following:
```
MESA_GL_VERSION_OVERRIDE="3.0 Mesa 17.2.5" python3 pycam/run_gui.py
```

sumpfralle, 2018-02-15:
@cki61: did you make progress? Can the issue be closed?

cki61, 2018-02-15:
Hi sumpfralle,
just tested the actual version, now I can see the canvas with some lines, but no model. Please see image appended.
Best Regards,
Christof
[image]

sumpfralle, 2018-02-16:
Hi Christof, 

could you please verify, that the model visualization is enabled?

[image]

cki61, 2018-02-16:
Hi sumpfralle,

I didn't recognize that "show Model" is a button, By your hint I saw it after pushing that area, it became blue in the background, see image Thanks a lot!

Best Regards,
Christof

[image]

closed 2018-02-16 by cki61