SebKuzminsky/pycam #99: TypeError

valeriob01, 2017-12-28:
Storing workspace in file: /home/sel/Downloads/newpycam/pycam/Aluminum1.yml
Traceback (most recent call last):
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/Project.py", line 318, in <lambda>
    ("SaveAsWorkspace", lambda widget: self.save_workspace_dialog())):
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/__init__.py", line 271, in save_workspace_dialog
    filename_templates=(self.last_workspace_uri, self.last_model_uri))
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/FilenameDialog.py", line 136, in get_filename_dialog
    dialog.set_current_name(os.path.basename(default_filename).encode("utf-8"))
TypeError: Must be string, not bytes
Traceback (most recent call last):
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/Project.py", line 318, in <lambda>
    ("SaveAsWorkspace", lambda widget: self.save_workspace_dialog())):
  File "/home/sel/Downloads/newpycam/pycam/pycam/Gui/__init__.py", line 271, in save_workspace_dialog
    filename_templates=(self.last_workspace_uri, self.last_model_uri))
  File "/home/sel/Downloads/newpycam/pycam/pycam/Plugins/FilenameDialog.py", line 136, in get_filename_dialog
    dialog.set_current_name(os.path.basename(default_filename).encode("utf-8"))
TypeError: Must be string, not bytes

sumpfralle, 2017-12-28:
Thanks!

Please check if this works now.

closed 2017-12-28 by sumpfralle

valeriob01, 2017-12-28:
Saving/loading workspace seem to work

commit 7dd742a0 (2017-12-28): filename dialog: fix str/bytes handling

Previously (python2) decoding was necessary.

Closes: #99