SebKuzminsky/pycam #105: Roadmap?

unrealsolver, 2018-01-02:
I haven't found any roadmap here. I would like to know if we have one? As a CNC owner and also software engineer, I am interested in improvements in PyCAM. At this moment I know following issues:
1. Better testing. including unit testing, some minimal performance testing, and, hopefully, cross-env testing.
2. Numpy integration.
3. Better UX/UI - requires an experienced person. I'm not talking about changing libraries, just move/change current UI to be more professional-looking.

sumpfralle, 2018-01-05:
> I haven't found any roadmap here. I would like to know if we have one?

I am not aware of a roadmap.
Discussions regarding plans for changes usually happen on the [devel-pycam mailinglist](https://sourceforge.net/p/pycam/mailman/). Please join this list, if you want to contribute or you are interested in discussions.

Personally I am working on the topics that I am interested in. Currently these are mainly:
* improvements of pycam's new non-interactive mode
* better testing
* integration of external libraries as replacements for internal algorithms of low quality (e.g. waterline/contour, polygon offsetting, ...)

Since your favourite topics do not overlap too much with mine, there seems to be plenty of space for exploring new terrain :)
And I guess, we can inspire each other in the field of tests.

I am looking forward to your thoughts or contributions!

valeriob01, 2018-01-12:
Integration with NumPy should boost up calculations and add transparent use of GPU...

closed 2018-02-08 by unrealsolver