SebKuzminsky/pycam #108: Cannot create process with release (0.6.2): list index out of range

jayMcBee, 2018-01-08:
When launching the app the command line states that the ProcessStrategies.py plugin is skipped due to broken or missing dependencies.

When trying to create a new process in the app a "list index out of range" error is printed on the Console and no process is added..

So the app is basically not usable on my system?
It's a fresh LinuxCNC install with just pycam-0.6.2 installed.

Any help appreciated.

valeriob01, 2018-01-08:
Had the same problem. Attempted to add the dependencies by hand to a LinuxCNC system, broke LinuxCNC update. Now I run pycam on debian and LinuxCNC on a separate machine.

sumpfralle, 2018-01-09:
@jayMcBee: could you share the full output during this error with us? I failed to see an obvious problem in ProcessStategies.py after taking a quick glance.

@valeriob01: I have the feeling that your issue is not related to jayMcBee's. And I hope that the recent switch of PyCAM to Python3 and GTK3 simplified the dependencies for you significantly?

jayMcBee, 2018-01-09:
Thanks for the quick response - unfortunately that was pretty much all in the Console, no line numbers, no stack trace, no additional info about which dependencies were missing.

I've tried the current pycam but apparently LinuxCNC uses an ancient Debian version not capable of running the support libraries required by the latest pycam dev version.

Eventually I downgraded to Pycam 0.6.1 which doesn't have the ProcessStrategies issue and works just fine (albeit it would be nice to use the *current* pycam with the *current* LinuxCNC, but that's probably an issue of the LinuxCNC team bundling a more modern kernel..)

SebKuzminsky, 2018-01-09:
Hi @jayMcBee , I realize we're getting a little side tracked here, but I want to let you know that we of the LinuxCNC project are testing Debian Stretch with RT-Preempt as the realtime platform for the next release.  As always no one knows how well that kernel will do on any particular hardware, but it might be interesting for you to try.  I've been running my bridgeport on it for a while now, it seems stable and realtimey enough on my setup (using Mesa hardware, not software stepping).

Try it and let us (LinuxCNC, not PyCAM!) know how it goes: http://linuxcnc.org/testing-stretch-rtpreempt/

jayMcBee, 2018-01-09:
Here we go - found the verbose option.
This is the output when starting 0.6.2 on my vanilla LinuxCNC install:

jaybee@cnc-jaybee:~/Downloads/pycam-0.6.2$ scripts/pycam -d
Enabled 2 parallel local processes
Skipping plugin /home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ProcessStrategies.py (broken or dependencies missing)
Skipping plugin /home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/PathPatterns.py (broken or dependencies missing)
Skipping plugin /home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/PathParameters.py (broken or dependencies missing)
Initializing module StatusManager (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/StatusManager.py)
No events registered for event 'plugin-list-changed'
Initializing module Units (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Units.py)
No events registered for event 'plugin-list-changed'
Initializing module ProgressBar (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ProgressBar.py)
No events registered for event 'plugin-list-changed'
Initializing module Toolpaths (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Toolpaths.py)
No events registered for event 'plugin-list-changed'
Initializing module PluginSelector (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/PluginSelector.py)
Initializing module Models (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Models.py)
Initializing module ModelSupport (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelSupport.py)
Initializing module ModelPlaneMirror (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelPlaneMirror.py)
Initializing module ToolpathCrop (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ToolpathCrop.py)
Initializing module ModelExtrusion (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelExtrusion.py)
Initializing module ModelScaling (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelScaling.py)
Initializing module FilenameDialog (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/FilenameDialog.py)
Initializing module ModelSupportDistributed (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelSupportDistributed.py)
No events registered for event 'visual-item-updated'
Initializing module ModelSwapAxes (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelSwapAxes.py)
Initializing module OpenGLWindow (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLWindow.py)
No events registered for event 'visualize-items'
No events registered for event 'visualize-items'
Initializing module ParameterGroupManager (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ParameterGroupManager.py)
Initializing module ParallelProcessing (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ParallelProcessing.py)
No events registered for event 'parallel-processing-changed'
Initializing module OpenGLViewAxes (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewAxes.py)
Initializing module ToolpathGrid (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ToolpathGrid.py)
Initializing module OpenGLViewGrid (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewGrid.py)
Initializing module ModelPosition (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelPosition.py)
Initializing module Clipboard (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Clipboard.py)
Initializing module ModelSupportGrid (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelSupportGrid.py)
Initializing module ModelProjection (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelProjection.py)
Initializing module ModelExport (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelExport.py)
Initializing module ModelExportContour (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelExport.py)
Initializing module ModelExportTrimesh (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelExport.py)
Initializing module ModelRotation (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelRotation.py)
Initializing module Fonts (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Fonts.py)
Font directory: /home/jaybee/Downloads/pycam-0.6.2/share/fonts
Initializing module OpenGLViewModel (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewModel.py)
Initializing module OpenGLViewModelGeneric (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewModel.py)
Initializing module OpenGLViewModelTriangle (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewModel.py)
Initializing module Tools (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Tools.py)
Initializing module OpenGLViewSupportModelPreview (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewSupportModelPreview.py)
Initializing module Processes (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Processes.py)
Initializing module OpenGLViewTool (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewTool.py)
Initializing module Bounds (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Bounds.py)
Initializing module ModelPolygons (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ModelPolygons.py)
Initializing module MemoryAnalyzer (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/MemoryAnalyzer.py)
Initializing module OpenGLViewBounds (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewBounds.py)
Initializing module Log (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/Log.py)
Initializing module OpenGLViewToolpath (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/OpenGLViewToolpath.py)
Initializing module ToolpathProcessors (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ToolpathProcessors.py)
[... cut ...]
Initializing module ToolTypeBallNose (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ToolTypes.py)
Initializing module ToolTypeBullNose (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ToolTypes.py)
Initializing module ToolTypeFlat (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ToolTypes.py)
Initializing module TaskTypeMilling (/home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/TaskTypes.py)
STL import info: 302857 / 538976266 / 0 / 23
Imported STL model: 0 vertices, 0 edges, 1444 triangles

sumpfralle, 2018-01-10:
Interesting!

The source of the problem seems to be this one:
```
Skipping plugin /home/jaybee/Downloads/pycam-0.6.2/pycam/Plugins/ProcessStrategies.py (broken or dependencies missing)
```

I just improved the log message for this situation a bit. This should help us determine the cause of the above event. Please upgrade your `stable/0.6` branch or apply the patch manually: 80a0a5e0136d979b8deffff21609b83833f9a114

Please share the new log message with us.

sumpfralle, 2018-02-15:
@jayMcBee: did you try it again with the improved logging to be found on the `stable/0.6` branch?

jayMcBee, 2018-02-16:
Sorry - the OS where this occurred is no longer installed here due to limited interop with Pycam..

sumpfralle, 2018-02-17:
Thank you for your quick response!

> Sorry - the OS where this occurred is no longer installed here [..]

What a pity!

Could someone else share more details regarding this issue? Otherwise I am tempted to close it, since I cannot reproduce it.

august9999, 2018-04-25:
...any help appreciated

sumpfralle, 2018-04-25:
Now I got it: you need to install the `python-enum34` package.
I just added this to the `stable/0.6` branch.

Thank you for your report!

closed 2018-04-25 by sumpfralle

august9999, 2018-04-26:
Thank you very much for your help!!!!!